thesourcerer8/altium2kicad #3: Problem with line segments, segment should be gr_line on F.Silk

nickoe, 2015-01-03:
The straight lines in the rounded rectange that is supposed to containg the chinese logo thing, they are written as segments, where they should not. I don't know how to fix this. I just traced the problem.

The problem is with line segments, segment should be gr_line on F.Silk. The line that is problematic is:
```
3866 #Tracks#528: 210C00FFFFFFFFFFFFFFFFFFFFAA486E036D9C21022A0299036D9C210294330100000000000000000006000301
3867   (segment (start 146.20039 -90.82320) (end 153.31239 -90.82320) (width 0.20000) (layer F.SilkS) (net 1))
```
If changing to 
```
3867   (gr_line (start 146.20039 -90.82320) (end 153.31239 -90.82320) (width 0.20000) (layer F.SilkS))
```
it is editable as a line in pcbnew.

This was done with ef52d4b5af9b174594c2d2786d235dda9647fa0f

nickoe, 2015-01-03:
Doing the above will fix a problem where the lines/segments are not rendered in the 3D viewer.

If I change line 1310 in convertpcb.pl to use `gr_line` instad of `segment` and remove `(net 1)`. Then it will render finde. I don't understand the code block around that section, but it also makes the silk on some footprints appear in the 3D viewer. I don't know if ti breaks anything else.

nickoe, 2015-01-03:
It seems like some other layers are also pased at that line, so maybe it would be usefull to enforce `gr_line` when it is `*.Silk` and a line that would otherwise be called a `segment`.

thesourcerer8, 2015-01-04:
Fixed in https://github.com/thesourcerer8/altium2kicad/commit/c55f58f3b5f245b9a5c0964dde97bb5d350b43a3

closed 2015-01-04 by thesourcerer8