thesourcerer8/altium2kicad #4: Power port labels are not translated

peabody124, 2015-05-12:
In my Altium schematic, the original file uses labeled power ports for different powers

[image]

Unfortunately these are lost in the translation:
[image]

Also if you look to the lower right all the parametric fields in the information block are lost and just put the values from altium without lookup: e.g. =sheetnumber

However, **AWESOME**work. This is amazing.

thesourcerer8, 2015-05-13:
Please try again whether the new version solved this problem.

peabody124, 2015-05-18:
Thank you very much! Much better.
[image]

It is nice to release OSHW in an OSS format.

thesourcerer8, 2015-05-18:
Thanks for your feedback!

closed 2015-05-18 by thesourcerer8