thesourcerer8/altium2kicad #5: Any way to integrate this into KiCad? Please!

timofonic, 2015-11-06:
Hello.

I would love if KiCad provides better interoperability. Think of it as the "Document Liberation Project" of EDAs.

Would you consider in helping them to develop an Altium importer?

Regards.

thesourcerer8, 2015-11-09:
Hello.

Sure, I am willing to help. Which help exactly do you need from me?

Do you have any specific Altium files or people who have Altium files in mind that you have as a usecase? 

I have been thinking a bit about it in the past, and I wasn´t sure, whether the converter should be 3 seperate converters for schematics, PCB layout and libraries and integrated into the respective KiCad modules, or whether it should be one project importer that imports the whole project at once, generating several PCB, schematic and libraries.

Another question is which quality level we want to reach with the converter.

Regards.

hildogjr, 2018-01-24:
@timofonic, you could propose this in the official KiCad development https://bugs.launchpad.net/kicad and see that the community think about.