thesourcerer8/altium2kicad #11: Do you have a way to unpack the altium's libs ?

diyjack, 2016-07-04:
What about convert altium's schlib and pcblib to kicad?

thesourcerer8, 2016-07-04:
I had started working on them too, but I have not found many library files on the internet. Can you provide me with files you want to have converted?

matthiasbock, 2016-07-13:
I have a couple of Altium libraries online:
* https://github.com/matthiasbock/Altium-Libraries

Also, I had some success parsing/unpacking Altium libraries using Python:
* https://github.com/matthiasbock/python-altium

usmG, 2017-07-10:
http://valhalla.s3.amazonaws.com/AD10-Libraries/Libraries.zip

some altium libraries that can be useful for you

a-v-s, 2018-11-07:
Is there any update on the possibility to convert PcbLib files?

eabase, 2019-09-20:
Bump!! 

Can someone please report back on the status of this? I'm only supplied Altium Designer Lib files for some quite complicated parts that I do not wish to try to recreate manually. At least not the pads...