thesourcerer8/altium2kicad #17: Database formats

sslupsky, 2017-01-04:
Hello,  I was wondering if you know the compatible Altium database formats for the converter?  I have an older version of Altium (6.9) from which we would like to convert some files to Kicad.  

I used the converter to convert the schematics.  There was a problem with the grid.  I determined that Kicad requires a 50mil grid and many of the components and wires imported were off grid.

I ran the convertpcb.pl scrip but I have been unable to get the pcb converted.  When I run the script I see quite a few errors.  I have attached some snips of the errors I noticed below.

Parsing error in Pads, header code 02 does not match 6A at pos 204

The following block of errors repeats several times:

Use of uninitialized value $aty in subtraction (-) at /altium2kicad/convertpcb.pl line 1728.
Use of uninitialized value $id in hash element at /altium2kicad/convertpcb.pl line 1733.
Use of uninitialized value $d{"MODELID"} in hash element at /altium2kicad/convertpcb.pl line 1749.
Use of uninitialized value in split at /altium2kicad/convertpcb.pl line 1755.
Use of uninitialized value $d{"MODEL.3D.ROTX"} in subtraction (-) at /altium2kicad/convertpcb.pl line 1758.
Use of uninitialized value $d{"MODEL.3D.ROTY"} in subtraction (-) at /altium2kicad/convertpcb.pl line 1758.
Use of uninitialized value $d{"MODEL.3D.ROTZ"} in subtraction (-) at /altium2kicad/convertpcb.pl line 1758.
Use of uninitialized value $id in hash element at /altium2kicad/convertpcb.pl line 1761.
Use of uninitialized value $id in hash element at /altium2kicad/convertpcb.pl line 1767.

There is one warning as well:

Warning: Pourindex -1 out of the expected range (0 .. 100)

And this is the last line:

Texts6...

I suspect my issue(s) maybe related to the vintage of my Altium files.

sslupsky, 2017-01-05:
Came across this Compound Document parser today.  The guy used it to parse Altium file into JSON.  Thought I would bring it to your attention.

https://github.com/dejan-stankovic/meetup-ad-schematic

thesourcerer8, 2017-01-05:
Unfortunately, it's hard to guess what the solution for the problem is just from your logs. The problem is that everyone has different habits of using Altium Designer which results in using different specialities in the files that I haven't seen in any of the other filess I have seen yet. It would be very helpful if you could send me your files.

sslupsky, 2017-01-05:
Sure, I could send you the files.  They are a bit sensitive so I would prefer not to post them in a public forum.  Could you let me know your email address?

thesourcerer8, 2017-01-05:
Thanks for the link, but only half of the files is textual inside, the other half is binary inside the Compound Document container. And even converting the textual parts to KiCad is challenging.

thesourcerer8, 2017-01-05:
Please upload them here: www2 . futureware . at / ftp /

sslupsky, 2017-01-05:
Ok, I uploaded the files.  Let me know if they are helpful.

thesourcerer8, 2017-01-05:
Thanks, I received 4 files. Both PCB and Schematics were convertable, but the results aren't good yet. I removed some of the warnings. Unfortunately the issues need deeper investigation. Could you perhaps provide a PDF of the schematics, and Gerber files of the PCB, and perhaps screenshots?

thesourcerer8, 2017-01-05:
I don't fully understand your first comment regarding the grid in KiCad. Could you explain this a bit more? I did not notice anything off-grid or any grid related problems. Could you perhaps send me a screenshot?
(Are you using a recent KiCad 4 version?) 
You can configure the grid in KiCad, did you find that option yet?

thesourcerer8, 2017-01-05:
convertpcb.pl generated a upload4.kicad_pcb properly for me, which can be opened in KiCad.

thesourcerer8, 2017-01-05:
The "Parsing error in Pads, header code 02 does not match 6A at pos 204" is a problem with parsing, somehow one record seems to be a few bytes shorter than my converter expected it to be. I'll have to compare it with various other test files I have to figure out how to decode this properly.
Regarding "Warning: Pourindex -1 out of the expected range (0 .. 100) " : As far as I remember, KiCad wants the priorities to be in the range 0..100, Altium sometimes uses 0..100 and sometimes 1000..1100. I had not seen -1 to occur in any of files I have seen yet, but it seems that there are other files on the internet which are POURINDEX=-1. I am not sure what the meaning of -1 could be here for the polygon priority. If means  just one below 0, then perhaps I should sort and renumber all occuring POURINDEX values, but perhaps -1 has another meaning. If you could open the file in Altium and make screenshots of the polygon settings, that might help.

thesourcerer8, 2017-01-05:
Regarding the libraries (.PcbLib and .SchLib): At the moment my converters can only convert .PcbDoc and .SchDoc files, the library handling part is only half finished, it parses them a bit but does not produce outputs yet. I guess that you can export the used footprints from the converted .PcbDoc and .SchDoc files, if you need them. But if you really need the .PcbLib and .SchLib converted, please tell me.

sslupsky, 2017-01-05:
Ok.  Just uploaded the pdf's.  I will dig up the gerber files and upload those as well.

sslupsky, 2017-01-05:
Uploaded the gerber files as well and a screen shot of the polygon pour screen.  Looked around a bit but could not find anything that seems like a reference to POURINDEX.  If you have a specific screenshot in mind, let me know and I can get it.

sslupsky, 2017-01-05:
When I used the script the first time (quite a few months ago), when I opened the files in Kicad a lot of the parts and wires were not on a 50 mil grid.  That creates a problem in Kicad because everything needs to be on a 50mil grid or you run into selection issues, wire end issues, etc.  The Kicad developers recommend the 50 mil grid (they even have a rule check in the library editor to check for this).

I will see if I still have those versions to get some screen shots.

thesourcerer8, 2017-01-05:
POURINDEX is the priority option of the polygons, which one is rendered first and which one is rendered later on. Perhaps they are "In the foreground" "Always in the foreground" "In the background" "Always in the background" or something like that.

sslupsky, 2017-01-05:
The Polygon Pour screenshot shows the order of the pours.  There are 4 pours.  Is there a fifth pour in the data?  Maybe it is an "end of pour" marker of some sort?

sslupsky, 2017-01-05:
Uploaded some annotated screenshots from Kicad.  These were taken using Kicad 4.0.5.  I ran the conversion using your scripts yesterday.  The results are very similar to what I recall I had seen a few months ago when I ran them.

thesourcerer8, 2017-01-05:
When I right-click on the Schema Editor, I have a menu option "Grid Select", where the default is 50 mil, but I can change it down to 1 mil there.

sslupsky, 2017-01-05:
Yes, that is correct.  Unfortunately, Kicad does not have a "snap to pin" function.  Thus, if you place a component on a 10 mil grid and try to place a wire to a component that is on a 50 mil grid, you have to use the 10 mil grid to ensure that the wire attaches to both components.  And if you want to select wires, you have to have set the grid.  So, either you use the "lowest common denominator" or you are always switching your grid back and forth.

This is not likely anything you can do anything about since it likely comes in this way from Altium.

sslupsky, 2017-01-05:
I uploaded a few screen shots of the Kicad PCB file that was written from the scripts.  It looks like there are only two "components" in the design.  And neither of those components, if you select them, appears to be connected to their primitives.  I think I recall the flag is one of the two components.

There are no pads and no net names.

thesourcerer8, 2017-01-05:
Ok, I fixed the 50 mil alignment now.

sslupsky, 2017-01-06:
Really?  Gee, that was unexpected, thanks!

thesourcerer8, 2017-05-21:
Could you please check now, whether there are still issues with your file that you want solved?

closed 2018-06-01 by thesourcerer8