thesourcerer8/altium2kicad #20: webench export to Altium

rhowlett, 2017-03-03:
I'm trying to convert a webench design into kicad.   Webench (https;//webench.ti.com) is a Texas Instruments tool for power supplies and what not. They don't support exporting directly into kicad, but they do support Altium.  What webench spits out doesn't follow what your code is looking for.  I've never used Altium so I'm not sure what to expect for it's output.  Webench generates a zip file with the following files:

~/Altium/2017-03-02_19-43-15$ll
total 760
-rw-rw-r-- 1 rhowlett rhowlett  31200 Aug  2  2013 UL_Import.PrjScr
-rw-rw-r-- 1 rhowlett rhowlett    263 Aug  2  2013 UL_Form.pas
-rw-rw-r-- 1 rhowlett rhowlett    983 Aug  2  2013 UL_Form.dfm
-rw-rw-r-- 1 rhowlett rhowlett  36096 Jan 29  2016 UL_Import.pas
-rw-rw-r-- 1 rhowlett rhowlett   3296 Apr  1  2016 readme.txt
-rw-rw-r-- 1 rhowlett rhowlett  37412 Mar  2 19:43 2017-03-02_19-43-15.txt
-rw-rw-r-- 1 rhowlett rhowlett    757 Mar  2 19:43 2017-03-02_19-43-15.prjdoc
-rw-rw-r-- 1 rhowlett rhowlett  77725 Mar  2 19:43 2017-03-02_19-43-151.schdoc
-rw-rw-r-- 1 rhowlett rhowlett  50280 Mar  2 19:43 0603.stp
-rw-rw-r-- 1 rhowlett rhowlett 404924 Mar  2 19:43 DR74.stp
-rw-rw-r-- 1 rhowlett rhowlett  50372 Mar  2 19:43 0402.stp

In the code, I see you are looking for a file called FileHeader.dat in a directory call Root Entry.  It looks like you are stuffing the contents of that file into a var called $contents.  Just for grins do you think it would be possible to add an example Altium design (sch and pcb) so I have something to step though the code with so I know exactly what should be in the dat file?  I write a bit of perl so this shouldn't be too much of a challenge:)

The convertpcb.pl gets a bit further but generates no files, just an empty directory (I'm going to look at that next).

thesourcerer8, 2017-03-03:
You have to unpack the .schdoc and .pcbdoc files with the "unpack.pl" first, it will give you the FileHeader.dat and the Root Entry directory. Those are inside the .SchDoc and .PcbDoc files. 
On the startpage (https://github.com/thesourcerer8/altium2kicad) I wrote: 
"To convert your Altium project to KiCad:
Go to the directory with your .PcbDoc and .SchDoc files and run:
    unpack.pl (it unpacks the .PcbDoc and .SchDoc files into subdirectores)"
Is that not clear enough? Do you have any ideas how I could improve those instructions? Should I add some warnings to the convertpcb.pl and convertschema.pl that you have to run unpack.pl first if they don't find anything? Or should they do it automatically?
In case this does not work for you, can you send me your files? I will take a look at Webench and try to make sure the files they generate import properly.

rhowlett, 2017-03-03:
Sorry, thought I did that...  Let me try again.

rhowlett, 2017-03-03:
~/Altium/2017-03-02_19-43-15$unpack.pl 
Done.
~/Altium/2017-03-02_19-43-15$ll
total 760
-rw-rw-r-- 1 rhowlett rhowlett  31200 Aug  2  2013 UL_Import.PrjScr
-rw-rw-r-- 1 rhowlett rhowlett    263 Aug  2  2013 UL_Form.pas
-rw-rw-r-- 1 rhowlett rhowlett    983 Aug  2  2013 UL_Form.dfm
-rw-rw-r-- 1 rhowlett rhowlett  36096 Jan 29  2016 UL_Import.pas
-rw-rw-r-- 1 rhowlett rhowlett   3296 Apr  1  2016 readme.txt
-rw-rw-r-- 1 rhowlett rhowlett  37412 Mar  2 19:43 2017-03-02_19-43-15.txt
-rw-rw-r-- 1 rhowlett rhowlett    757 Mar  2 19:43 2017-03-02_19-43-15.prjdoc
-rw-rw-r-- 1 rhowlett rhowlett  77725 Mar  2 19:43 2017-03-02_19-43-151.schdoc
-rw-rw-r-- 1 rhowlett rhowlett  50280 Mar  2 19:43 0603.stp
-rw-rw-r-- 1 rhowlett rhowlett 404924 Mar  2 19:43 DR74.stp
-rw-rw-r-- 1 rhowlett rhowlett  50372 Mar  2 19:43 0402.stp
drwxrwxr-x 2 rhowlett rhowlett   4096 Mar  2 20:56 .
drwxrwxr-x 4 rhowlett rhowlett   4096 Mar  2 21:32 ..
~/Altium/2017-03-02_19-43-15$

No new directory(s)...?

rhowlett, 2017-03-03:
ok, looked at the script.. need to give it an argument (that could be clearer).  So now I get a directory, but it is empty.  Looking at the unpack.pl script now:)

rhowlett, 2017-03-03:
so, I started to see a bunch of unpack functions and then a line with:
next unless($revision==62); # We might stumble across other ASCII files that way
So, I looked at the file and who would have guessed, its an ASCII file!  Oh well...  actually looks like I could write a script for this.  Do you know if anyone is tackling this?  Its a:
|HEADER=Protel for Windows - Schematic Capture Ascii File Version 5.0

rhowlett, 2017-03-03:
Well I'm on my way!
https://github.com/vadmium/python-altium/blob/master/format.md

thesourcerer8, 2017-03-03:
Oh, that seems to be Schema in the old Protel 5.0 format. I only ever found one example of it, but I didn't tried to support it yet, since I believed it's too old that there will be much demand for it. 
Ok, for this you definitely do not need the unpack.pl, which is only for the newer Altium Designer formats, but the old format isn't packed.
I created a modified version of the convertschema.pl (the new protel branch) now which parses the file directly, but it somehow does not convert it correctly, and I don't see why at the moment.
To work, you have to create a 2 subdirectories "A" and "A/Root Entry" and copy the *.SchDoc into a file named "A/Root\ Entry/FileHeader.dat" (I'll change it to parse them directly, without the subdirectories in the future).
I don't have an idea why it does not parse correctly.
Yes, vadmium documented the schematic file-format.

rhowlett, 2017-03-04:
cool!

thesourcerer8, 2017-05-21:
Did it work for you? Do you need any further improvements?  Should I merge it into the official altium2kicad branch?

cezarsantanna, 2017-07-21:
I tried to use the Protel branch, but all the symbols are all together in one point, all the wires stay on the original position, the same with text, only the symbols are wrong!

thesourcerer8, 2018-01-05:
I have merged the Protel branch back into the master branch, and improved the parsing of the Webench generated files.