thesourcerer8/altium2kicad #21: IO error opening converted kicad_pcb

cscheidl, 2017-04-14:
Dear Sorcerer,
Your convertpcb.pl creates a kicad_pcb out of a Altium PcbDoc. But when trying to open it in Kicad an io-error is reported: Error loading board.
IO_ERROR: Maximum line length exceeded
from /tmp/buildd/kicad-0.20140622+bzr4027/common/richio.cpp : ReadLine() : line 183
I will FTP the PcbDoc File to You. Would be great, if You find some time to look at it.
Your work is appreciated very much. Thanks.

thesourcerer8, 2017-04-14:
Ok, I have an easy workaround for you:
Change $annotate=1 to $annotate=0 on line 39 in convertpcb.pl, this disables including the debugging information in the generated file, and solves this scalability problem of KiCad. Please also report it to KiCad, I think they should also fix such scalability issues. 
I am not sure, whether the converted file is right, I see some texts that look mostly good (some strange overlapping that does not look good), and I only see a very tiny PCB which also looks somehow broken. Please let me know if there are further issues I should take a look at, and perhaps provide some Screenshots, PDF or Gerber Files that I can check against.

cscheidl, 2017-04-15:
Thanks for the hint. After defining the layers at the top of the kicad_pcb with the text editor, the tiny pcb appeared in Kicad. I wanted to start with something very tiny...
For the moment, I don't need further assistance.

closed 2017-04-15 by cscheidl