thesourcerer8/altium2kicad #22: Missing Pads in the PCB file.

sheldonj84, 2017-04-23:
Ran your script as instructed but it seems all the pads in the layout file are missing. Is this a known issue. Is there a step I need to follow thats not in the instructions before the new pcb file will match up?

Thanks for your help

thesourcerer8, 2017-04-23:
No, I that's not a known issue, pads are usually converted correctly. Do you get any error messages during the conversion? The only known issue are octagonal pads, which are currently converted to rectangles, and they could overlap or be too small, since KiCad does not support octagonal pads. But you should get a warning message for each octagonal pad.  Can you send me the .PcbDoc and Gerber Files for testing? You could reduce the file to just a few items if you want.

ytwuyv, 2017-05-19:
I met the same problem,there is no pads in the kicad_pcb file which was converted from a PcbDoc file.
[P024D120A 0523 BIN-PcbDoc.kicad_pcb.zip](https://github.com/thesourcerer8/altium2kicad/files/1014609/P024D120A.0523.BIN-PcbDoc.kicad_pcb.zip)
Please try to convert it ,and I want to know what is wrong.

Thanks a lot.

ytwuyv, 2017-05-19:
Oh,I forgot to upload the source PcbDoc file ,it's here.
[P024D120A 0523 BIN.PcbDoc.zip](https://github.com/thesourcerer8/altium2kicad/files/1014620/P024D120A.0523.BIN.PcbDoc.zip)

gillen, 2017-06-13:
Same issue as "sheldonj84" but I've an information to add. When I tried in 19Feb2017 same procedure it worked (pads converted correctly) now I have the same problem as "sheldonj84". The log write: Parsing error in Pads, header code 02 does not match 1F at pos 856.

LOG:
c:\altium2kicad-master>unpack.pl
Loading ALIM_CTRL-2.PcbDoc

Done.

c:\altium2kicad-master>convertpcb.pl
Handling ALIM_CTRL-2-PcbDoc/Root Entry/Board6/Data.dat
ASCII-ALIM_CTRL-2-PcbDoc.PcbDoc exists?
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Board6/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Rules6/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Classes6/Data.dat.txt
Writing PCB to ALIM_CTRL-2-PcbDoc.kicad_pcb
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Nets6/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Components6/Data.dat.txt
Parsing error in Pads, header code 02 does not match 1F at pos 856
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Components6/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Arcs6/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Vias6/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Polygons6/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Tracks6/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/FileVersionInfo/Data.dat.txt
Writing to ALIM_CTRL-2-PcbDoc/Root Entry/Regions6/Data.dat.txt
Texts6...

Thank You

thesourcerer8, 2017-06-13:
Could you please send me the file? Which Altium Version is it from? You could also try the online conversion service: http://www2.futureware.at/KiCad/

gillen, 2017-06-14:
I tried the online version with same result, attached files below. Altium version is 16.0.5
Thank You

Files uploaded and downloaded to online version:
[ALIM_CTRL-2.zip](https://github.com/thesourcerer8/altium2kicad/files/1073789/ALIM_CTRL-2.zip)
[altium2kicad.zip](https://github.com/thesourcerer8/altium2kicad/files/1073790/altium2kicad.zip)

File previously converted in right manner:
[Alim_ctrl.zip](https://github.com/thesourcerer8/altium2kicad/files/1073803/Alim_ctrl.zip)

gillen, 2017-06-14:
I forgot to say that the files were originally created with Orcad 10.5 and then imported with  Altium wizard.
Thank You

thesourcerer8, 2017-06-20:
I updated convertpcb.pl now to fix your pad issues. It isn't perfect yet, but it's better.

gillen, 2017-06-20:
Unfortunately I do not see any changes, the board does not show any pads as before. Any suggestions?
Thank you for your work anyway
gillen

thesourcerer8, 2017-06-21:
I would say the import from Orcad into Altium did not work properly, the pads are not connected to the components anymore, as far as I can see. KiCad does not support component-less pads, so I changed the converter to create a "DEFAULT" component for component-less pads.
Do you still have the original Orcad files?

gillen, 2017-06-21:
Here is the original orcad file(*.Max) + Library (*.LLB), but I wonder why in February did the conversion work without modifications?
Thank You
[Orcad to Altium.zip](https://github.com/thesourcerer8/altium2kicad/files/1093004/Orcad.to.Altium.zip)

thesourcerer8, 2017-06-22:
I think that altium2kicad was never before able to cope with component-less pads. Perhaps something changed in your Altium version? If you need the pads to be properly connected to the components, please make sure that the Altium wizard does that, you could contact Altium to fix that for you. Developing an Orcad->KiCad seems to have a similar complexity than the altium2kicad project.

gillen, 2017-06-25:
Hello, after several attempts I rebuilt the way I was able to convert orcad files to kicad. This is the procedure: with Altium wizard import the orcad files then export to the Pcad-Ascii format, then launch Pcbnew of kicad but not through the kicad launcher then open the Pcad-Ascii suffix file and the board is now also complete of the pads.
Thanks for your efforts to help me.

thesourcerer8, 2017-07-04:
Have you also tried the latest altium2kicad verison?

gillen, 2017-07-05:
Hello, i tried a few minutes doing the new version, it works !!!! Thank's for your job.
gillen

closed 2017-10-13 by thesourcerer8