thesourcerer8/altium2kicad #23: no unpacked PcbDoc files found

usmG, 2017-07-10:
Hello i have used online altium to kicad converter 
I am  getting the same file for .schdoc and .pcbdoc in converted zip file
below is the conversion log 
please help i confused what is problem
version of altium file that need to be taken care or any version can be converted ?

Archive:  db/249.zip
   creating: db/249/test/
  inflating: db/249/test/SMPS.SchDoc  
  inflating: db/249/test/SMPS.PcbDoc  
Done.
No unpacked .PcbDoc documents found. Trying to unpack it:
Done.
There were no unpacked PcbDoc files found.

thesourcerer8, 2017-07-10:
Please move them out of the test directory, just put them directly into a .zip file. I will try to update the service later to improve subdirectory handling.

usmG, 2017-07-10:
it i working , but the pads are missing , in entire board i can see only one pad.

Thank's for your job.

thesourcerer8, 2017-07-10:
I changed it now to remove the subdirectories from the zip files, and I updated it to the newest version, it seems I forgot to deploy the newest version. Please try again now.

thesourcerer8, 2018-01-05:
The pad handling might be improved now, please try again.

DreV96, 2018-03-14:
[image]

I get the same problem. When I use the altium2kicad converter it automatically downloads a .zipfile without any .pro files in it.

thesourcerer8, 2018-03-16:
The ZIP file you have uploaded contains only Gerber files, no Altium Design files.
You can open Gerber Files in GerbView, which is part of KiCad. When you have opened them in GerbView you can convert them to a .kicad_pcb , and then continue with PcbNew to work on them, but the Gerber files do not contain all the information that was in the Design files, so the preferred way is to convert .PcbDoc design files instead of Gerber files.

closed 2018-06-01 by thesourcerer8