thesourcerer8/altium2kicad #24: No .pro file after conversion

ArjenLux, 2017-08-08:
Hi.
I tried converting 2 projects, but in both cases there was no .pro file in the output zip..
in the conversion log i dont see any error's, only warnings.

Writing to control-PcbDoc/Root Entry/Polygons6/Data.dat.txt
Warning: Pourindex -1 out of the expected range (0 .. 100)
Warning: Pourindex -1 out of the expected range (0 .. 100)

Any idea what im doing wrong?

Regards...

thesourcerer8, 2017-08-10:
I am sorry, altium2kicad currently does not generate .pro files yet. I usually use inside KiCad "File->New Project->New Project", then select the directory where you unpacked the files, write a filename and click on Save. Then KiCad asks whether it should create a new subdirectory, click on No. That's it. Should I add .pro file generation to altium2kicad?

nickoe, 2018-09-29:
I think altium2kicad do create pro files now. I guess this can be closed.