thesourcerer8/altium2kicad #25: Converting from Protel design files

BrendanSimon, 2017-09-06:
Can `altium2kicad` convert from `Protel` design files?

If not, is there another tool to convert from `Protel`?

Maybe an intermediate tool, that possibly `altium2kicad` can use?

Thanks, Brendan.

thesourcerer8, 2017-09-07:
I created a Protel branch for altium2kicad, if you use that, it should work with Protel files. If it doesn't work for you, please send me the file.

BrendanSimon, 2017-09-08:
I run the `unpack.pl` script on my `.Sch` and `.PCB` files.  It seems to complete without error (return code is `0`), but nothing is generated except for an empty directory.  I tried running the script without arguments too.

**Am I running the script correctly?**

The only two source files I have are: `Display Module - RevC01.PCB` and `Display Module.Sch`.

If I run `strings <file> | grep -i protel` I get the following.

`DProtel for Windows - Schematic Capture Binary File Version 1.2 - 2.0`

`PCB 4.0 Binary FileP>S:\UserTempArea\FredBloggs\Protel99\Display Module$RN>Display Module.DDB$OP>$ON>acmeinc.dot$ID>1$ATTR>0$E>Sch$STF>`

**Are these suitable for conversion with `altium2kicad`?**

thesourcerer8, 2017-09-08:
The unpack.pl is only needed for the newer Altium files, not for the older Protel files. Try the convertschema.pl and convertpcb.pl instead please. I adapted the convertschema.pl now to read the .Sch files from the current directory, and generate ...-kicad.sch files from it. I am not sure, whether the .PCB files can be converted, since the format has possibly changed a lot.
To answer your previous question: An intermediate tool to convert from Protel to KiCad would be a Altium Designer or CircuitMaker.

BrendanSimon, 2017-09-10:
Ok.  Got a little further in the sense that a `...-kicad.sch` and `...-kicad-cache.lib` file was generated, but opening the `.sch` shows an empty sheet.

The contents of the `.sch` show a number `LIBS:`, `ELAYER:`, lines, etc, but there aren't any coordinates for anything.

There was some output to `stdout` which may have been intended for file output?

```
Handling Display_Module.sch
Weiter
LINENO -> 0
LINENO -> 1
...
LINENO -> 147
LINENO -> 148
?*r?*h?*^?*T?*J?*@?*6?*????$?R?*$?H?*$?>?*$?4?*$?*?*$? ?*$??*$?
                                                               ?*????$HR?*$HH?*$H>?*$H4?*$H*?*$H ?*$H?*$H
                                                                                                         ?*????$?R?*$?H?*$?>?*$?4?*$?*?*$? ?*$??*$?
                                                                                                                                                   ?*????$?
                                                                                                                                                           ?*$??*$???*$???*$???*$???*$???*$???*????$H
                                                                                                                                                                                                     ?*$H?*$H??*$H??*$H??*$H??*$H??*$H??*????$?
                          ?*$??*$???*$???*$???*$???*$???*$???*????$???*$???*$???*$???*$???*$???*$???*$???*????$H??*$H??*$H??*$H??*$H??*$H??*$H??*$H??*????$???*$???*$???*$???*$???*$???*$???*$???*????$(N?*$(D?*$(:?*$(0?*$(&?*$(?*$(?*$?*????$????*$????*$????*$????*$????*$????*$????*$????*????$???*$???*$???*$???*$???*$???*$???*$???*????$???*$???*$???*$?r?*$?I?*$? ?*$????*$????*????$x??*$x??*$x??*$x??*$x??*$x??*$x??*$x??*????$???*$???*$???*$?r?*$?I?*$? ?*$????*$????*????$???*$???*$???*$???*$???*$???*$???*$???*????$:??*$:??*$:??*$:r?*$:I?*$: ?*$:???*$:???*????$,??*$,??*$,??*$,??*$,??*$,??*$,??*$,??*????$???*$???*$???*$?r?*$?I?*$? ?*$????*$????*????$???*$???*$???*$???*$???*$???*$???*$???*????$,?*$,V?*$,-?*$,?*$,???*$,???*$,???*$,`??*????$x?*$n?*$d?*$Z?*$P?*$F?*$<?*$2?*????$??*$?V?*$?-?*$??*$????*$????*$????*$?`??*????$xx?*$xn?*$xd?*$xZ?*$xP?*$xF?*$x<?*$x2?*????$??*$?V?*$?-?*$??*$????*$????*$????*$?`??*????$?x?*$?n?*$?d?*$?Z?*$?P?*$?F?*$?<?*$?2?*????$:?*$:V?*$:-?*$:?*$:???*$:???*$:???*$:`??*????$,x?*$,n?*$,d?*$,Z?*$,P?*$,F?*$,<?*$,2?*????$??*$?V?*$?-?*$??*$????*$????*$????*$?`??*????$?x?*$?n?*$?d?*$?Z?*$?P?*$?F?*$?<?*$?2?*????$??*$????*$????*$????*$????*$????*$????*$????*????$????*$????*$????*$????*$????*$????*$????*$?~??*????&?*?*??*??*??*??*??*????$-?*$#?*$?*$?*$?*$??*$??*$??*????$@}?*$@s?*$@i?*$@_?*$@U?*$@K?*$@A?*$@7?*????$?$?*$?*$??*$?	?*$??*$???*$???*$???*????$???*$???*$???*$???*$???*$???*$???*$???*????$W??*$W??*$W??*$W??*$W??*$W}?*$Wt?*$Wk?*?"?$??*$??*$??*$??*$??*$x?*$o?*$f?*????$???*$???*$???*$???*$???*$???*$???*$???*????${?*${??*${??*${??*${??*${??*${??*${??*????$???*$?w?*$?m?*$?c?*$?Y?*$?O?*$?E?*$?;?*????$?j?*$?a?*$?X?*$?O?*$?F?*$? -> ?*$?4?*$?+?*????$???*$???*$???*$???*$???*$???*$???*$???*????$\j?*$\a?*$\X?*$\O?*$\F?*$\=?*$\4?*$\+?*????$
```

thesourcerer8, 2017-09-10:
No, the KiCad output is only textual, not binary, so it displays some of the binary input from the file, but it isn't  supposed to go to the file output.
I am sorry, altium2kicad currently cannot parse your files. If possible, please provide your files so that we can improve the converter.

Engineer8468, 2020-09-17:
Hello, I hope these messages are still being read!  I have some old Protel99SE projects I'd like to convert to Kicad.  It was mentioned that there was a branch designed for Protel, but it doesn't appear to be present here.  Is it available/working?