thesourcerer8/altium2kicad #26: Flexible layers

sinaAESD, 2017-09-15:
Sorry not an issue but a general question which I did not know where to ask.

Does the code also takes into account the different materials for different layers? I have flexible layers connecting some small islands and the main board.

Thanks in advance

thesourcerer8, 2017-09-16:
As far as I know, KiCad does not specifically support flexible layers and different materials for different layers yet, the way Altium does it. So those informations will get lost during the conversion. So I would suggest that you try to convert it and see what is missing. Then please try to see whether it is possible to design any missing information in KiCad, and tell me, then I will try to add those to the converter. If it is not possible to design it, and you really need it, please send feature-requests to KiCad.
I have seen PCB designs in the past, with a graphical description of the layer stackup and the materials and thicknesses to be used, next to the PCB (outside the printable range), so that the factory knows what they have to do.

sinaAESD, 2017-10-06:
Thank you, will investigate the converted version and keep you updated.