thesourcerer8/altium2kicad #28: Description of internal structure of PCBDOC file

kalmuk, 2017-09-26:
Hi,
Is there any documentation which describes how to read files inside PCBDOC. Many files are strings which are coded binary data and I don't know how to read each number. I can dive into code for altium2kicad converter, but maybe there is some official documentation provided by Altium (I wrote an email to their support but I'm waiting for an answer). 
Anyone heard about this or have some documentation?

thesourcerer8, 2017-09-26:
Those are 2 places where I got a few pieces of information about the fileformat from, but mostly about the schematics, not the PCB:
https://github.com/upverter/schematic-file-converter/blob/master/upconvert/parser/altium.py
https://github.com/vadmium/python-altium
I haven't heard or seen any official documentation from Altium about it, and I would be surprised if it existed. If you find or get anything, please share it if possible.

kalmuk, 2017-09-26:
Thank You for a very quick answer, but I can't find there any helpful information. My problem is that I want to get components map from PCBDOC file (using some script). Do You know where is data about component designator? In which file after unpacking PCBDOC I can find it. As I said before I'm diving into Your code, but I cant get exact position of this data. 
For each component I need: 
- designator
- x_min (minimal horizontal position of given component/designator on PCB)
- x_max (maximal ...)
- y_min (minimal vertical ...)
- y_max (maximal ...)

thesourcerer8, 2017-09-26:
This information does not exist in an easy form in the PCBDoc files. And also not in the ASCII files from Altium, as far as I know. In "Root Entry/Components6/Data.dat" you have the fields X and Y coordinates, where the component is placed. The record number is the Component number. In  "Root Entry/ComponentBodies6/Data.dat" you have MODEL.2D.X and MODEL.2D.Y. Then there are the Pads, the Vias and the 3D Models... I think you have to parse through all of them, simulate placing them on a PCB and calculate the min/max yourself. Perhaps you can get those values from Pick&Place files that were generated by Altium? If that does not work out, I guess that you will need a high data quality from altium2kicad. altium2kicad currently still seldomly places a few things in the wrong place, which is not a big problem for a user that looks at the design, but will likely cause wrong min/max values for your application. Which min/max do you actually need? The one of the pads, or the one of the physical package of the component in 3D? How accurate does it need to be?

kalmuk, 2017-09-26:
Actually I need coordinates for designator in which component will be placed. I need to create map of all designators. I know how to do it via ASCII file (sample component info from ASCII is below), but all projects in my company is in PCBDOC, so in this case I have to convert it via Altium to ASCII and then parse it, I need some automatization.
Pick&place file have only center of component, I need the frame in which component is placed.

I've found in Your code how to get lines from 4th to 9th from the sample of ASCII below (and this is center of component). Below lines with CT is info about lines of designator (description on example) and below CP lines is info about component pads. 
I think that I can find info about Pads in Your code, but it can be inaccurate. The most valuable for me is info below CT line, do You know where can I find it? I don't need 3D model.

> COMP
CON2SIG050
0 0 18429995 9374951 1 0 0 0 0 0 0 270.000 1 1 16
CS
0 0 18437330 9400590 16000 90.000 1 2 18 0 0 0 0 2000 1 0 0
X365
CS
0 0 18473892 9374951 28634 90.000 1 2 18 0 0 0 0 2863 1 0 0
con2sig050_x_1
CT
0 0 18419864 <-line_x_beginning 9349077 <-line_y_beginning 18419864 <-line_x_ending 9400259 <-line_y_ending 3937 18 0 0 0
 0 0 1 0 0
CT
0 0 18419864 9400259 18439549 9400259 3937 18 0 0 0
 0 0 1 0 0
CT
0 0 18439549 9349077 18439549 9400259 3937 18 0 0 0
 0 0 1 0 0
CT
0 0 18419864 9349077 18439549 9349077 3937 18 0 0 0
 0 0 1 0 0
CP
0 0 18429715 9387461 0 0 1 0 0 1 25600 19700 2 0 0 16 0 0 90.000 0 -16735 10000 10000 20000 4 1 1 1 1 1 1 1 1
1
CP
0 0 18429715 9361871 0 0 1 0 0 1 25600 19700 2 0 0 16 0 0 90.000 0 -16735 10000 10000 20000 4 1 1 1 1 1 1 1 1
2
ENDCOMP

thesourcerer8, 2017-10-05:
I am sorry, but the COMP/CT/ENCOMP block you have posted does not look familiar to me. Please let us know when you find any information or documentation about the fileformats.