thesourcerer8/altium2kicad #30: Error loading PCB file

tnogueira, 2017-10-04:
Hi,

First of all thanks for the great tool! I've generated a .kicad_pcb file from an Altium PCB using your online tool. In the log i cannot really see any error, but when i open the .kicad_pcb in KiCad i get the following error:
------------------
Error loading board.
PARSE_ERROR: Expecting "(" in input/source '3001510x-02_A_CC6UL_Starter_Board-PcbDoc.kicad_pcb'
line 8437
offset 41
from dsnlexer.cpp : Expecting() : line 361
------------------

Attached follows the .kicad_pcb as a .txt file. Any help is very welcome.

Thank you

[3001510x-02_A_CC6UL_Starter_Board-PcbDoc.txt](https://github.com/thesourcerer8/altium2kicad/files/1355879/3001510x-02_A_CC6UL_Starter_Board-PcbDoc.txt)

thesourcerer8, 2017-10-05:
I have solved the problem. Please try again now.

tnogueira, 2017-10-05:
Thanks for the quick reply. 

It did solve the problem for this particular file. When i try with a different one, it fails with the same error

Error loading board.
PARSE_ERROR: Expecting "(" in input/source '3001517x-03_A-PcbDoc.kicad_pcb'
line 1053
offset 24
from dsnlexer.cpp : Expecting() : line 361

I attach the original and the converted file. Thanks!

[3001517x-03_A-PcbDoc.kicad_pcb.txt](https://github.com/thesourcerer8/altium2kicad/files/1360622/3001517x-03_A-PcbDoc.kicad_pcb.txt)
[3001517x-03_A.PcbDoc.txt](https://github.com/thesourcerer8/altium2kicad/files/1360623/3001517x-03_A.PcbDoc.txt)

thesourcerer8, 2017-10-05:
That's another problem that I have solved too now.

tnogueira, 2017-10-05:
Great, it works now.
Thanks!

closed 2017-10-05 by tnogueira