thesourcerer8/altium2kicad #34: Uninitialized value

gearhead, 2017-11-16:
I tried this script tonight and get this:
Handling Sheet1-SchDoc/Root Entry/FileHeader.dat
Use of uninitialized value in multiplication (*) at ./convertschema.pl line 110.
Use of uninitialized value in multiplication (*) at ./convertschema.pl line 111.
Use of uninitialized value in multiplication (*) at ./convertschema.pl line 110.
Use of uninitialized value in multiplication (*) at ./convertschema.pl line 111.

I do not need the board, per se, but the schematics would be great. As it is, I get an incomplete schematic and a board outline and no traces. 

In the code for convertschema.pl, it looks for imagemagick from a dos/windows  directory structure:
`my $searchimagemagick="\\Program Files\\ImageMagick-6.8.9-Q16\\";`
Is this correct? Am I doing something wrong?

thesourcerer8, 2017-11-16:
The warning messages indicate that some values that it expected do not exist in the file you provided. Could you send me your Altium schematic files for investigation?

The ImageMagick is necessary if the schematic contains images, so that they can be converted from BMP format which Altium uses to PNG which KiCad uses. If your schematic does not contain images, you don't need to care about that path. If your schematic contains images, you should install ImageMagick and put the correct path into the script there.

gearhead, 2017-11-16:
I cannot send the schematic, unfortunately. 

I have ImageMagick installed and put my path as "/usr/bin/" I get different results when I put this path in.
[image]

This is a snippet of one of the resulting schematics.

thesourcerer8, 2017-11-17:
Are you using the most current version? The lines you have warnings in are currently commented out.
I guess that you cannot share your schematic for secrecy reasons. If the problem persists: Please try to delete half of the schematic in Altium, and try again. If the problem goes away, try to delete the other half. If the problem is still there, delete another quarter of the schematic, ... do it until you can narrow down where the problem is, and hopefully you will get to a schematic that you can share with me. If still necessary, I could sign an NDA.