thesourcerer8/altium2kicad #35: How to parse the Data.dat.txt of Pads6?

7Chen, 2017-11-16:
Hi~ 
I want to get some information about pads from Pads6, but it's format is unreadable and different from the Data of Components6, how can I parse the format as follow, which extract from Pads\Data.dat.txt 

0202000000 A:        0131 010000000005000000047C267C300100000000AA000000010C00FFFFFFFF7700FFFFFFFF674F7702CEB78102AF9B0600BC9A0300AF9B0600BC9A0300AF9B0600BC9A03000000000002020200000000008056400000000000000000A08601000400A0860100400D0300400D030000000000409C0000000000010101010101010000000000000000000001000001000000409C00 00006F2B48E334C02044AE0AA4D7EC7AFCAC2473142206AE25499A6A9A9BB9AEAB8F00000000FFFFFF7FFFFFFF7F00000000

Thanks in advance.

thesourcerer8, 2017-11-16:
Which information do you need about the pads? If you figure anything out, please let me know. Will the software you are developing be OpenSource? I would be interested in it.

7Chen, 2017-11-17:
Coordinate, size and type of pads, and  the relationship between pads and components is also my interest.
I use Altium, it is not opensource. What puzzled me is the coding mode, it seems like combined ascii and binary data, Altium hided the details.

thesourcerer8, 2018-06-05:
I have improved the Pads parsing a lot lately. convertpcb.pl currently outputs a file Pads.html where it writes an annotated hex-dump of the pads binary data. It marks all fields it understand in yellow, and when you move the mouse over the yellow fields, it explains which field it is. 
A short explanation: At first every pad starts with the record-type byte, which is always 0x02, then there are 6 fields, that all have a 4-byte length field and then the contents of those fields. The first field contains the name of the pad, and the 5th field contains the interesting data, coordinates, size and type of the pad.