thesourcerer8/altium2kicad #36: Failure to convert multiple sheet project

SchrodingersGat, 2017-11-16:
Hi @thesourcerer8 this is a great tool! I am trying to convert a complex Altium project and the output is nearly identical! This is a ten-layer board with ~300 components, so very impressive :)

One issue I have found is that it does not seem to work for multiple sheet hierarchies. 

* Subsheets are not embedded
* Hierarchical labels are not imported

Here are some screenshots:

**Altium:** 

[image]

**KiCad:**

[image]

**Altium:**

[image]

**KiCad:**

[image]

**Altium:**

[image]

**KiCad:**

[image]

Any thoughts or tips on how to approach this issue?

thesourcerer8, 2017-11-16:
Could you please try to generate a very small test-case for me in Altium with such a multiple hierarchy, then please convert it to KiCad, then copy the KiCad and edit it to be correctly hierarchical in KiCad. Then please send me the Altium files, the converted KiCad files and the corrected KiCad files for inspection. Please also send screenshots, and please explain how the hierarchy works exactly, so that I can analyze that properly.

IJeffray, 2018-01-22:
I have just added support for ports, sheet entries and crude support for harnesses.  Perhaps try your conversion again with the latest code?  It's far from perfect as I don't have a huge number of examples to work with so if you're able to share some that'd be great.  [I'd be happy to handle the data under NDA if required]