thesourcerer8/altium2kicad #37: Pads conversion failure with PCBDoc from AD17, AD15.  Simple test case.

toybuilder, 2017-12-01:
I'm not sure if it's some setting that is an issue, but boards that I create in AD17.1 (or even testing with AD15) will fail during conversion of pads.

Boiling down the problem to its essence, [this simple board (one component with just a single pad)](https://www.dropbox.com/s/7lmuqvlw41l4ku5/a2k-failure-case-1.PcbDoc?dl=0) and also [this other one (one component with a single pad; and a separate free pad)](https://www.dropbox.com/s/19adknnbmpdd3k3/a2k-failure-case-2.PcbDoc?dl=0), one free pad) shows the problem behavior.

There's clearly some kind of extended data stored with the pads list, which the current parsing is not handling well:

`
Handling a2k-failure-case-1-PcbDoc/Root Entry/Board6/Data.dat
Writing to a2k-failure-case-1-PcbDoc/Root Entry/Board6/Data.dat.txt
[...]
Pads6...
Warning: This is a rounded rectangle pad, and those are currently not supported by KiCad. We convert them to rounded pads for now, please verify the PCB design afterwards. This can cause overlaps and production problems!
substr outside of string at /cygdrive/c/Users/Joseph/Dropbox/work/safecast/test-altium/altium2kicad-master/altium2kicad/convertpcb.pl line 98.
substr outside of string at /cygdrive/c/Users/Joseph/Dropbox/work/safecast/test-altium/altium2kicad-master/altium2kicad/convertpcb.pl line 1746.
Use of uninitialized value in string eq at /cygdrive/c/Users/Joseph/Dropbox/work/safecast/test-altium/altium2kicad-master/altium2kicad/convertpcb.pl line 1746.
substr outside of string at /cygdrive/c/Users/Joseph/Dropbox/work/safecast/test-altium/altium2kicad-master/altium2kicad/convertpcb.pl line 1791.
Use of uninitialized value in string ne at /cygdrive/c/Users/Joseph/Dropbox/work/safecast/test-altium/altium2kicad-master/altium2kicad/convertpcb.pl line 1791.
Writing to a2k-failure-case-1-PcbDoc/Root Entry/Components6/Data.dat.txt
Writing to a2k-failure-case-1-PcbDoc/Root Entry/Tracks6/Data.dat.txt
Writing to a2k-failure-case-1-PcbDoc/Root Entry/FileVersionInfo/Data.dat.txt
Texts6...

Handling a2k-failure-case-2-PcbDoc/Root Entry/Board6/Data.dat
Writing to a2k-failure-case-2-PcbDoc/Root Entry/Board6/Data.dat.txt
Writing to a2k-failure-case-2-PcbDoc/Root Entry/UniqueIDPrimitiveInformation/Data.dat.txt
[...]
Pads6...
Warning: This is a rounded rectangle pad, and those are currently not supported by KiCad. We convert them to rounded pads for now, please verify the PCB design afterwards. This can cause overlaps and production problems!
Parsing error in Pads, header code 02 does not match 05 at pos 205 (0xCD)
Writing to a2k-failure-case-2-PcbDoc/Root Entry/Components6/Data.dat.txt
Writing to a2k-failure-case-2-PcbDoc/Root Entry/Tracks6/Data.dat.txt
Writing to a2k-failure-case-2-PcbDoc/Root Entry/FileVersionInfo/Data.dat.txt
Texts6...
`

thesourcerer8, 2017-12-10:
I have improved the pad parsing somewhat, removed some of the warnings, and added a heuristical try to continue parsing if it cannot parse the file correctly yet. Regarding the rounded rectangles, please support KiCad to add support for rounded rectangles (and octagonals), so that we can convert them.

thesourcerer8, 2018-06-01:
The problem has been fixed.

closed 2018-06-01 by thesourcerer8