thesourcerer8/altium2kicad #41: Error loading board. IO_ERROR: Maximum line length exceeded from richio.cpp : ReadLine() : line 235

sabiha1233, 2018-02-28:
Hi,
I have converted a altium design file to kicad using altium2kikcad by following the video [](https://www.youtube.com/watch?v=fDEzrdIp9xA). When I tried to open the converted board file I got "Error loading board. IO_ERROR: Maximum line length exceeded from richio.cpp : ReadLine() : line 235". I found a discussion related to this error. Reference link: [](https://forum.kicad.info/t/pcbnew-max-file-size/2383/12) In this link, the suggested solution is to remove the unnecessary data to make long lines shorter in kicad_pcb file. My file(Please find the attached file) has 27223 and I have no idea which lines to make shorter. Please help me in this regard.
[board.zip](https://github.com/thesourcerer8/altium2kicad/files/1765687/board.zip)

Thank You.
[board.zip](https://github.com/thesourcerer8/altium2kicad/files/1765690/board.zip)

thesourcerer8, 2018-02-28:
Please open the file convertpcb.pl , there is on line 39 the command $annotate=1; please change that to $annotate=0; then run it again. I will try to fix this problem permanently soon.

thesourcerer8, 2018-02-28:
I have added linebreaks in the new version of convertpcb.pl, please try this new version, and tell me whether it solves your problem.

sabiha1233, 2018-03-03:
Hi, I tried both (Changing $annotate=1 to $annotate=0 and downloaded the new version) but both failed to open the converted file. I am attaching the output I got in terminal while the convertpcb.pl is running. And when i tried to open the file I got error "PcbDoc.kicad_pcb', line x: '=' expected." 
Please help me in this regard.
[image]
[image]
[image]
[image]
[image]

sabiha1233, 2018-03-06:
Hi,I have tried solutions you provided and updated the status. Please check
and help me to solve the issue

thesourcerer8, 2018-03-16:
Can you please provide the original .pcbdoc file?

thesourcerer8, 2018-06-05:
I have changed the default behaviour to not do the annoations anymore, could you please try again, and if it doesn't work, can you please send me the original .pcbdoc file?