thesourcerer8/altium2kicad #43: XMOS audio board schematic altium2kicad not working?

xlordofpainx, 2018-04-04:
Hey guys, I am pretty new here, but I would like to ask you if you can help me with converting a schematic to KiCad as I am unable to manage that. This is the link to the zip file https://www.xmos.com/support/boards?product=18334&component=19663
Thank you in advance.

evanshultz, 2018-05-02:
I didn't do any checking, but using Strawberry Perl 5.26.2.1 I was able to convert this design. I've zipped the resulting files and attached them for you.
[xCORE-200-Multichannel-Audio-Platform-KiCad.zip](https://github.com/thesourcerer8/altium2kicad/files/1965544/xCORE-200-Multichannel-Audio-Platform-KiCad.zip)

thesourcerer8, 2018-06-05:
I also didn't noticed any problems with the conversion. Could you explain what the problem was, what you tried?