thesourcerer8/altium2kicad #47: ERROR: Wrong recordtype: 0600, expected 0100 at pos 0

laf0rge, 2018-06-13:
I've attempted to conver the OpenVizsla 3.2 Altium design from https://github.com/openvizsla/ov_ftdi/tree/master/hardware but I get the following error message:

`
ERROR: Wrong recordtype: 0600, expected 0100 at pos 0.
This record type is really unknown. Please contact the developer to add it.
`

This is not super critical for me, I just wanted to give this converter a try and see how the results look like.  As it's an OSHW design, you can try debugging the conversion error, if you're interested. Thanks!

gregsmart, 2018-06-19:
Hi,
I had a go at this for you, and I had the same issues I've had with my own boards.  The pcb works fine except that all the vias are missing.  I get an error loading the schematics, but it's easy to fix.  I haven't checked, but I suspect it is due to a format change in Kicad 5.0.0-rc2.
You haven't provided any information about what you did to get this error.  It's impossible to reproduce if you don't include the steps you took.
The generated PCB is included.

BTW. This converter has got to the point where it is quite useful and there aren't that many issues to sort out.

[ov_ftdi-PcbDoc.kicad_pcb.zip](https://github.com/thesourcerer8/altium2kicad/files/2113874/ov_ftdi-PcbDoc.kicad_pcb.zip)

laf0rge, 2018-06-21:
Thanks a lot for your help, @gregsmart.

I simply used the online converter at http://www2.futureware.at/KiCad/ to do the conversion, there were no specifics steps I had to take.  Maybe that online converter is using a different version than the latest master here in the git repo?

thesourcerer8, 2018-06-21:
Sorry, I forgot to update the online converter lately, I did that now. I am planning to take a deeper look at the provided file in the next few days.