thesourcerer8/altium2kicad #49: Document not converter, no .pro file to be found

VedantKalbag, 2018-08-26:
I've uploaded all the documents that I could get, which was the .schdoc, the .pcbdoc, and the BOM for the project in an excel sheet, but when I unzipped the file after downloading, there was no .pro file to open. When I tried opening the .schdoc files with kicad, it opens, but only shows me the connections I've made, but not a single component. No ICs, resistors, capacitors, or even ground connections. Please help.

thesourcerer8, 2018-09-03:
Yes, at the moment, altium2kicad does not generate a .pro file, since it's currently 3 seperate tools, the unpacker, the schematics converter and the pcb converter, since all 3 are completely different file formats to be handled. Perhaps I should add a fourth tool that runs the 3 other tools and generates the .pro file? My personal workaround is to use KiCad to generate the .pro file, by doing "File->New Project->New Project", then going to the directory of the project, entering a project name and clicking. "Save". Then KiCad asks whether to create a new Directory, which I refuse, and that created the .pro file.

thesourcerer8, 2018-09-03:
Regarding the problem with the missing components: Could you provide your .schdoc files?