thesourcerer8/altium2kicad #58: Altium2kicad does not generate archives

BrianSantillan, 2018-11-19:
Hello there,
I am using Windows 10, Kicad 5 and Altium 17. I have downloaded Pearl to run the scripts. 
Is it possible to convert just the PCBDoc from Altium? Or do I need pcb and sch? I share my pcb.
I cant convert the file to kicad.
Thank you so much.
Regards.-

[DPM7107.zip](https://github.com/thesourcerer8/altium2kicad/files/2593442/DPM7107.zip)

thesourcerer8, 2018-11-19:
I would suggest to try to convert the .PCB file found in the DPM7107.zip with Altium 17 to the newer .PcbDoc file format, and afterwards convert it with the altium2kicad converter to KiCad format.

BrianSantillan, 2018-11-19:
Oh yes, Ive forgotten that. I have some Pads and Footprints missing. Is it normal? Nonetheless, we have a progress. I share one photo.
Thank you for helping me. 
Regards.
[image]

BrianSantillan, 2018-11-19:
Edit: Footprints are no missing, just Pads.