thesourcerer8/altium2kicad #66: Rounded rectangle SMD pads are imported as oval pads

cdwijs, 2018-12-04:
I have used SMD resisters in Altium that are made from Rounded Rectangle. These are incorrectly imported as oval pads. Changing the oval pads in Kicad to "Rounded Rectangles" pads corrects them.

cdwijs, 2018-12-05:
I was unable to upload my zip file with the Altium file, the Kicad file and the screenshot. therefore I've put it here, it can be downloaded within 30 days:
https://ufile.io/417ig

cdwijs, 2018-12-05:
In the directory Root Entry/Pads6 there are 4 files: Data.dat Data.dat.bin Header.dat Header.dat.bin The file Data.dat.txt is missing. This means I don't have a list of properties in the Altium file about the pads.
Why is this file missing?
How can I generate it?

And the real question I am trying to answer: What variable holds the data that I need to create a rounded rectangle pad in Kicad? The variable missing is roundrect_rratio:
(pad 2 smd roundrect (at 0 -1.125 90) (size 1.15 1.3) (layers F.Cu F.Paste F.Mask) (roundrect_rratio 0.25))

Cheers,
Cedric

cdwijs, 2018-12-05:
I've made an Altium file with pads with Rounded Rectangle with various Corner Radius. I can't see any change between the pads in the file pads.txt. Time to dig deeper :-)

rusefillc, 2021-01-02:
I see that something was merged here but we are also affected by this issue.

[image]

@andreika-git can you attach our file?

rusefillc, 2021-01-03:
Correction: my bad, ^^^ was produced NOT with latest version of altium2kicad

rusefillc, 2021-01-03:
[image]

looks better. why is this ticket still open @cdwijs @thesourcerer8 ?

thesourcerer8, 2021-01-08:
I could need some help with triaging the open tickets. I will close this ticket now, in case there are still issues, please reopen it.

closed 2021-01-08 by thesourcerer8