thesourcerer8/altium2kicad #72: convertschema.pl works but no .sch file produced

tyyao, 2019-04-15:
Hello altium2kicad creators and github community,

I ran the unpack.pl script and unpacked my .schdoc file into subdirectories. When I ran convertschema.pl, it did not convert the files in my subdirectories to .sch files. 
The screenshots below show the scripts being called to run in the command prompt and the subdirectories unpack.pl produced:
[image]
[image]

Could anyone please help me figure out if there is something I am doing incorrectly?

Many thanks,
tyyao

thesourcerer8, 2019-04-15:
You have to run both commands in the same directory. convertschema.pl automatically looks for those subdirectories.

tyyao, 2019-04-16:
Thank you so much thesourcerer8! (I like your name. It's cool) That did the trick, and now I have success converting altium schematics to kicad schematics.

closed 2019-04-24 by thesourcerer8