thesourcerer8/altium2kicad #75: The compressed file format is unknown or the data is corrupted

769898178, 2019-08-05:
Hi,this is the first time I use this function and I met a problem.After I update the zip document,what I download is a 0B zip document and when I try to open it ,it says"The compressed file format is unknown or the data is corrupted". I tried different PCB documents but the same result.I have no idea if I choose the wrong document,can you help me plz?
By the way what you do is really meaningful!
[PCB.zip](https://github.com/thesourcerer8/altium2kicad/files/3465724/PCB.zip)
[JY61.zip](https://github.com/thesourcerer8/altium2kicad/files/3465725/JY61.zip)
PS:What I really want to translate is only JY61.zip,an MPU6050 module.PCB.zip is only for test.

thesourcerer8, 2019-08-05:
The JY61.zip contains an old Altium/Protel 5.0 fileformat PCB. Can you open and save it with a newer Altium version, and then try to convert it again?

769898178, 2019-08-05:
> The JY61.zip contains an old Altium/Protel 5.0 fileformat PCB. Can you open and save it with a newer Altium version, and then try to convert it again?

Thank you for your reply.
Well,I tried again and found that the problem came from the downloader,it's alright after I used Google Chrome to download .But when I tried to translate PCB zip,I found many modules are still missing instead a box with ??.Maybe some modules are not compatible or my AD verion is too low(I still use AD 16)?

thesourcerer8, 2019-08-05:
AD 16 should be fine. Can you open it, and save it in the new fileformat of AD16 please, and try that again? And please post the file you upload here is well if it doesn't work.

769898178, 2019-08-05:
Still fail o(╥﹏╥)o
Kicad report this :
This schematic currently USES the engineering symbol library table to find the method of loading library symbols. KiCad will attempt to map existing symbols to use the new symbol library table. Remap will change some engineering files, and the schematic may not be compatible with older versions of KiCad. If you need to restore any changes, all changed files are backed up to the "remap_backup "folder in the project folder. If you choose to skip this step, you will be responsible for remapping symbols manually.
[Gyro.zip](https://github.com/thesourcerer8/altium2kicad/files/3467572/Gyro.zip)

thesourcerer8, 2019-08-05:
This file converted nicely, you should have a Gyro-SchDoc.sch and Gyro-SchDoc-cache.lib. Unfortunately the symbols are named "Cap" which collides with the symbols in the newer KiCad version, so KiCad suggests to replace them with the new internal version from KiCad. But you should be able to choose to update or to keep/rename them the way they are, and it should be fine.

thesourcerer8, 2019-08-05:
[GyroConverted.zip](https://github.com/thesourcerer8/altium2kicad/files/3467856/GyroConverted.zip)

769898178, 2019-08-05:
> This file converted nicely, you should have a Gyro-SchDoc.sch and Gyro-SchDoc-cache.lib. Unfortunately the symbols are named "Cap" which collides with the symbols in the newer KiCad version, so [...]

Understand. Thank you for your reply and  the function you developed!