thesourcerer8/altium2kicad #80: XML parsing error: 'syntax error' at line 1

ShaneKent, 2020-01-19:
I've tried to use altium2kicad on my project a few different times and I haven't had any success on the schematic conversion.   I can convert the PCB perfectly and it appears to be opening correctly in KiCAD.  Unfortunately, I can't seem to convert the schematic properly.  When I run the 'convertschema.pl' script, I get the following outputs:
`Handling PCB-PcbDoc/Root Entry/FileHeader.dat
Handling SHEET1-SchDoc/Root Entry/FileHeader.dat
Handling SHEET1.SchDoc`  and then nothing else.   

When I open the .sch file that is created I get an warning that says:
`Error loading schematic "............", 
Unable to read file "............"`
If I press OK then I'm met with an error saying:
`XML parsing error: 'syntax error' at line 1`
Nothing happens after clicking OK on that window.

It would be great if we could work together to try and solve this issue.
I'm running KiCAD 5.1.5 and perl v5.30.1.

I can send you the schematic file that is causing problems if you'd like that.

Thanks in advance!

thesourcerer8, 2020-01-19:
Please provide the schematic file.

ShaneKent, 2020-01-19:
Thanks for the quick response! I can't upload here on Github, so I've uploaded it to Dropbox here: https://www.dropbox.com/s/n0deoaeorymidjp/SHEET1.SchDoc?dl=0

ShaneKent, 2020-01-20:
@thesourcerer8 Can I be of any more support in trying to solve this issue?   This is gating my progress in a couple of projects so I'd like to help you close this however I can.

thesourcerer8, 2020-01-21:
I converted your file with convertschema.pl, and then I was able to open the file with the command "eeschema SHEET1-SchDoc.sch" without those errors in EESchema 4.0.7. I attached you the packed converted file, so that you can try it with my file. Please try whether you can open the file I attached, and I will install a newer KiCad version and try that one.
[SHEET1-SchDoc.sch.zip](https://github.com/thesourcerer8/altium2kicad/files/4093473/SHEET1-SchDoc.sch.zip)

thesourcerer8, 2020-01-21:
How exactly did you try to open the .sch file in KiCad?

ShaneKent, 2020-01-21:
@thesourcerer8 Thanks for sending this my way!  Perhaps I'm doing something wrong... I open Eeschema by opening just any old KiCAD schematic and then I try to go to 'Import Non KiCad Schematic' and it fails there.     Even with the zipped file you sent me it's still failing.

ShaneKent, 2020-01-21:
Ah! After a bit of research it seems as though I've figured out what I was doing wrong.    I can now open the schematics!! Thank you for your help, I think I was being a bit dense...

All of my library symbols are mapping over correctly except the power symbols (5V, GND, etc.) are there any tips or suggestions you have to make that import work?

thesourcerer8, 2020-01-22:
Kicad has several different GND symbols, and I am not sure whether the 5V symbol it has is the right one for you. I guess you should take a look which symbols you want/need exactly, and then you could exchange them manually or with a small script. Or we could see whether the mapping is generic enough that we could implement it into convertschema.

thesourcerer8, 2020-03-17:
I have updated the converter to a newer KiCad fileformat which solved the symbol issues for me, since KiCad does not try to convert them from the older version anymore. Does it help for you?

ShaneKent, 2020-03-18:
Yes! I just tried your newest update and that appears to have fixed the issue!  Thanks for that!

-Shane

closed 2020-03-18 by ShaneKent