thesourcerer8/altium2kicad #81: Does the converter actually work or what am I doing wrong?

epsilon-phi, 2020-02-13:
Hi,

first up - sorry for the bold question. I have been trying for some time to get the converter running but I am only getting bad results. I am new to github, so I hope this question is accepatble.

I am trying to convert a project that my university desperatly wants to have as a KiCad file. Trying to run the converter with it yields the following issues:
- The schematic is completely empty after saying is uses an old form of library list.
- In the layout the planes are there but only the perimeter is visible and it does not want to be filled in even after forcing with B.
- In the layout the component pads are circular instead of a rounded rectangle (which can be fixed by replacing to "rect" with a text editor)
- In the layout the vias have a drill diameter of 0.6ish instead of 0.3mm.

I have created a very small test project to hopefully solve these issues on my side myself but I am only getting the same issues, see the prictures for an explanation of the aforementioned points. Please find all my files attached. [TestPrj.zip](https://github.com/thesourcerer8/altium2kicad/files/4201651/TestPrj.zip) If it helps, I can gladly provide the big project as well.

[image]
[image]
[image]
[image]

Could it be a version issue, so that the converter would only work with specific versions of Altium and KiCad, juding from the "old form of library list" message? Which versions of Altium and KiCad was this converter verified with?

I am still hoping to not have to redo my project entirely in KiCad again. Help is greatly appreciated.

Best regards
Phil

thesourcerer8, 2020-02-28:
The fileformat of Alitum has changed, the schematic format has changed the parameter names from uppercase to mixedcase.
For me the filling works when i press "B".
The rectangular pads are a bit strange, perhaps the fileformat changed there as well, because the file seems to say that those have to be circles or ovals. Perhaps it is an oval with very sharp corners? Ovals are converted to circles by my converter at the moment. Can you please provide a test-file which contains all kinds of different shapes, together with screenshot how they look like, and perhaps text next to them that explains how they should look like? rectangular, oval, circle, ...
Can you please explain the drill diameter with larger screenshots and marking them? I am not used to drill diameters.

thesourcerer8, 2020-02-28:
Could you please correct the drill diameter in the KiCad file and send me both original and corrected file?

epsilon-phi, 2020-03-03:
Okay sure, I'll try my best.

So here is the test-file:
2D Altium:
[image]
2D KiCad:
[image]
3D Altium:
[image]
It seems that octagonal and rounded rectangle pads are both getting converted into round (=circle) pads. Slots are being converted to rounds (fair enough). For some strange reasons NPTH pads are not selectable in KiCad and they are shown only halfway through in the 3D view. 
Altium drills:
[image]
KiCad drills:
[image]
As for the vias, same picture as before: where the drill diameter should be 0.3, its 0.6 (mm). The via pad however is correct (see pictures). I discovered that the KiCad via description lines simply miss "(drill 0.3)". Maybe this will already fix the issue.
[image]
Here is a zip will all the files including pictures: [a2k-shapestest.zip](https://github.com/thesourcerer8/altium2kicad/files/4282889/a2k-shapestest.zip)

Do you have any ideas on the schematic yet?

thesourcerer8, 2020-03-05:
Ok, I have updated the convertschema.pl to handle the new file format. Thanks a lot for the test-files! I have added a rounded rectangle handling now, since KiCad is able to handle them now too. Can Altium define the percentage of roundedness? Could you please provide a test-file with various different "regions"? Triangles, rectangular, ...   and please have a bit more space between the different objects. I still have to investigate the drill issue.

epsilon-phi, 2020-03-05:
Oh this is great news. Thank you so much for your support, I appreciate it a lot!

Yes, Altium can define the percentage of roundness. Together with some regions and polygons, pads with different "roundnesses" are included in the new test file: 
[a2k-test2.zip](https://github.com/thesourcerer8/altium2kicad/files/4294790/a2k-test2.zip)
In Altium you see:
[image]
or in 3D:
[image]
In KiCad after the conversion you find:
[image]

I'll stay curious about how all this will come out. I'll also gladly provide more test data, if you should require so.

epsilon-phi, 2020-03-05:
After checking you latest revision of the schematic converter I found it looks much better now. All the Footprints now get imported correctly and the traces and sheet entries and even the references to other sheets are there. Only the Power & Ground Net-Symbols are still missing and being shown as ??-Boxes, however thats nothing you cant fix by hand. Thanks for the amazing work!
[image]

epsilon-phi, 2020-03-05:
> Only the Power & Ground Net-Symbols are still missing and being shown as ??-Boxes, however thats nothing you cant fix by hand.

I just re-read on the main page here, that this limitation is known. In that case, it seems like the converter does now do exactly what it should do. :D

thesourcerer8, 2020-03-06:
For me it worked once to just save the file, close pcbnew and open it again, and the symbols were there. Perhaps the converter does something wrong in the fileformat, I'll investigate ...

thesourcerer8, 2020-03-06:
Ok, I updated to the new kicad schematic fileformat, which solved the power symbols now.

thesourcerer8, 2020-03-06:
I added handling for the rectangle roundings now.

thesourcerer8, 2020-03-06:
The regions are converting to zones now.

thesourcerer8, 2020-03-06:
I removed the paste layer now which covered the holes

epsilon-phi, 2020-03-07:
Okay I tested it today and my power symbols are still missing. But I rather belive that this has to do with my KiCad Version or something like this as I just cant get the polygons - which are there - to fill (<- definitely a KiCad error).
As my original problem, that the converter did not do anything at all, has been solved now, I will close this thread and investigate a bit further on my side. Thanks a lot for your effort to adjust all these details in the process. Keep up the amazing work! Greetings

closed 2020-03-07 by epsilon-phi

pointhi, 2020-03-28:
@epsilon-phi 

Thanks for your testfiles. I tried them with my importer which will be soon integrated into KiCad:

* https://gitlab.com/kicad/code/kicad/-/merge_requests/60

It seems most things work, except: rotated slot hole,  rectangular hole and rounded zones. But those three things are not supported in KiCad and would need to be emulated anyways.

[image]
[image]