thesourcerer8/altium2kicad #82: Missing Hierarchical Label and polygons translated.

AnnaCosty, 2020-03-17:
Hi,
I have tried converting a project from Altium20 to Kicad 5.1.5 and I noticed two things
- all the hierarchical labels/pins were converted as global ones missing the connections between sheets.
- the polygon in the layout starts from a corner which is outside the actual pcb. This did not happen in the previous version of the tool.

Anyone else having same issues? how did you solve them?

thesourcerer8, 2020-03-17:
Do you mean the previous version of Altium or the previous version of the converter? Could you provide test-files please?

AnnaCosty, 2020-03-26:
Sorry for my late reply. I meant the prevision version of the converter tool. 
Attached you find the test file with the 2 errors: 
1. the ports in the sheets are translated as "GLabel" and this break the connections between the sheets. It should be instead  "HLabel" and in the main sheet it should also be part of the sub-sheet object.
2. the layout error can't be replicated fully but it affects all filled areas that have too many vertices, so in the test files you can see how a random image on the silk screen get translated.

I used Altium 20, Kicad 5.1.5
[test_files.zip](https://github.com/thesourcerer8/altium2kicad/files/4388438/test_files.zip)