thesourcerer8/altium2kicad #83: Broken .kicad_pcb created - Broken/Empty Zones added

FifoIronton, 2020-03-24:
I have been trying to convert [this open source project](https://github.com/foxBMS/foxbms/blob/master/hardware/hw-slave-18-ltc6813-1/hw-slave-18-ltc6813-1.zip) to KiCad from Altium. When converting, it appears that the Schematics come out _alright_ (each sheet has obvious errors, but nothing that can't be handled in a few hours), but the PCB is unable to be opened.  
Kicad gives the following error:
```
Expecting "(" in input/source 'C:\Users\Owner\Downloads\hw-slave-18\Sources\slave-PcbDoc.kicad_pcb'
line 38953
offset 17
```
When running convertpcb.pl I see many errors that look like this: 
```
Error: substring out of range: length:1524 pos:312812 wantedsize:8 field:Y19550 (main,/mnt/e/Software/altium2kicad/convertpcb.pl:2951 - main::msubstr)
Use of uninitialized value in unpack at /mnt/e/altium2kicad/convertpcb.pl line 2951.
Use of uninitialized value in division (/) at /mnt/e/altium2kicad/convertpcb.pl line 2951.
```

Broken zones in the resulting .kicad_pcb file look like this:
```
# Verts:                                                   
(zone  (layer ) (tstamp 547BA6E6) (hatch edge 0.508)       
    (connect_pads thru_hole_only (clearance 0.09144))      
    (fill (mode segment) (arc_segments 32) )               
    (polygon                                               
      (pts                                                 
      )                                                    
    )                                                      
  )                                                        
```

Manually removing these zones from the .kicad_pcb allows me to open the file, which appears to have some very strange zones and some broken silkscreen.

Outputs from each script:
[unpack.pl](https://pastebin.com/NyUmzvGF)
[convertschema.pl](https://pastebin.com/BfmN1RSM)
[convertpcb.pl](https://pastebin.com/pcTwTx8e)

spiccinini, 2020-05-16:
The problem seems to be related to the detection of the layers for the zones

rusefillc, 2021-01-10:
is this related to https://github.com/thesourcerer8/altium2kicad/commit/b330c5086880e629593b4201aca85920395aafe4?
is this related to #86?