thesourcerer8/altium2kicad #84: broken schematic

ltarmcool, 2020-03-31:
Hi, I downloaded the altium2kicad scripts and I was able to generated kicad schematic from the altium files. However, I don't see any lib files in the .cache library for kicad and the kicad generated schematic is all messed up with so many question marks. (I used strawberry pearl to run the scripts)

ltarmcool, 2020-04-01:
[image]

ltarmcool, 2020-04-01:
the kicad schematic looks like this 
[image]
while it's supposed to look like this
[image]

CxZMoE, 2020-08-04:
Me too..It seems altium2kicad doesn't convert the SchLib/PcbLib files. What a shame,and I need to switch back to my windows workflow.