thesourcerer8/altium2kicad #86: Faulty zones in converted project

telkamp, 2020-05-04:
I've converted the evaluation board STM32MP157A-DK1 (6 layer PCB) to KiCad, using the most recent version of altium2kicad converter. The altium project files are available online: 
[mb1272_bdp.zip](https://www.st.com/resource/en/board_manufacturing_specification/mb1272_bdp.zip)

In the redulting *.kicad_pcb file, I figured out the following problems:
- I'm getting a 99MB large log output, mostly with messages like:
Use of uninitialized value in division (/) at ../convertpcb.pl line 2950.
Error: substring out of range: length:500 pos:508 wantedsize:8 field:Y31 (main,../convertpcb.pl:2951 - main::msubstr)
Use of uninitialized value in unpack at ../convertpcb.pl line 2951.
- a net name of attached to some zone nets is not supported by KiCad 5.1.5. E.g. the zone defined in line 58251: zone (net 142 "SAI2_SCKA"). The net name "SAI2_SCKA" causes KiCad to give me an error message like... ")" expected. I have removed the net name attached to all zone nets in the *.kicad_pcb file
- some polygons have a number of xy positions (xy 0 0) at their end, e.g. in line 58481. The result is a zone that has a corner at the lowest left position of the pcb. I have removed all those positions, but it seems that now there are corners missing in some zones. Maybe the 0,0-positions are a result of the "uninitialized value in division (/)" problem
- maybe this problem has the same cause:
The resulting *.kicad_pcb file contains zones without any corners, e.g.
```
# Verts: -32512
(zone  (layer ) (tstamp 547BA6E6) (hatch edge 0.508)
    (connect_pads thru_hole_only (clearance 0.09144))
    (fill (mode segment) (arc_segments 32) )
    (polygon
      (pts
      )
    )
  )
```
I have to remove those zones.

rusefillc, 2021-01-02:
We are also affected.

@thesourcerer8 did you have a chance to look into this? My blunt question is - do you care? Is your preference for the community to look into this or would you be available? Can a donation attract your attention to this issue?

rusefillc, 2021-01-02:
@pantosaur @cdwijs @IJeffray is this something you might help with?

cdwijs, 2021-01-02:
KiCAD 5.99 is capable of importing Altium Files. Does that work for you?
https://forum.kicad.info/t/work-in-progress-native-altium-importer/20712/190

rusefillc, 2021-01-02:
@cdwijs 5.99? 5.1.9 maybe? let me try that!

By any chance do you know if KiCad offers this action as a command line tool?

cdwijs, 2021-01-02:
5.99 is not a typo. 5.1.9 does not know howto deal with Altium files. You need the Nightly Development Builds (v5.99) for that.
You can download them here for Windows:
https://kicad-downloads.s3.cern.ch/index.html?prefix=windows/nightly/

Use at your own risk. Make sure you have backed up your kicad files.

rusefillc, 2021-01-02:
@cdwijs are there plans to officially deprecate this https://github.com/thesourcerer8/altium2kicad/ project or are there any plans to keep this https://github.com/thesourcerer8/altium2kicad/ project alive? :) just if you know...

cdwijs, 2021-01-02:
No Idea. I'm not the maintainer of either altium2kicad nor Kicad itself. 

I just complain through bug reports if stuff does not work.

rusefillc, 2021-01-02:
thank you @cdwijs that's a lot of valuable info!

rusefillc, 2021-01-02:
Status update: looks like this issue is related to current version of Altium. We've saved our PCB in previous version format and conversion has worked.

rusefillc, 2021-01-03:
Correction: this is NOT about Altium version, this is about altium2kicad version.

With ~2018 version we have altium2kicad produce a valid kicad PCB but not in latest version. @andreika-git will attempt to isolate specific commit

andreika-git, 2021-01-03:
It's "Regions convert to zones now"  https://github.com/thesourcerer8/altium2kicad/commit/b330c5086880e629593b4201aca85920395aafe4

rusefillc, 2021-01-10:
@thesourcerer8 any chance for any acknowledgment or any transparency into the plans or lack of plans?

thesourcerer8, 2021-01-11:
Sorry for being silent so long. I thought that KiCad implementing its own Altium importer will reduce the need for altium2kicad, so I concentrated on other projects. It seems that there is still demand though, so I am currently trying to setup my own development and testing environment again (the old computers I was using aren't usable anymore) , so that I can develop, review and test new changes again. My current goals are: Fixing the open bugs, making it compatible to newer Altium versions, and finalizing the library support. Any help with prioritizing the issues, reviewing pull-requests, and developing improvements are highly welcome, and most likely needed to achieve the goals.

rusefillc, 2021-01-11:
@thesourcerer8 glad to hear from you :)

Your thinking makes total sense. While it looks like KiCad 6 is maybe just a few months away it would also be some time until everyone is ready to catch up with KiCad 6. For instance I have no idea when https://github.com/openscopeproject/InteractiveHtmlBom/ would be compatible with 6. Anyway 6 timing see https://www.reddit.com/r/KiCad/comments/iw2uez/when_kicad_v6_will_be_released/

Our reports of incompatibility with latest Altrium was a false alarm, it looks like it's latest https://github.com/thesourcerer8/altium2kicad which is affecting us. See message by @andreika-git above which points at when https://github.com/thesourcerer8/altium2kicad was broken for us. I would suggest to fix master as first priority? If that happens we are happy to report if zones/regions/etc would work as expected for us.

Please use your judgement. Worst case scenario we are the last use of this project who knows :)

thesourcerer8, 2021-01-11:
> It's "Regions convert to zones now" [b330c50](https://github.com/thesourcerer8/altium2kicad/commit/b330c5086880e629593b4201aca85920395aafe4)
Can you try to pinpoint which part of the change is causing the problem?

bunker-bunk567, 2021-02-01:
@rusefillc

the region through (0,0) is fixed with https://github.com/bunker-bunk567/altium2kicad/commit/b6a935d7c59cced3665b73746e7c5c79810a2676

bunker-bunk567, 2021-02-02:
> Sorry for being silent so long. I thought that KiCad implementing its own Altium importer will reduce the need for altium2kicad,

probably not. altium2kicad has a much lower barrier for contribution and much less requirement on code quality with respect to performance and integration into the whole. this would allow altium2kicad to be used as reference (very important for a converter for data where correctness matters) and to prototype features.

only thing it does not have is tighter access to kicad (library) functionality. kicad itself might be able to fix problems better.

thesourcerer8, 2021-02-02:
I agree, both approaches have their advantages and disadvantages.