thesourcerer8/altium2kicad #102: Various problems importing complex pcb

albmac, 2021-01-12:
Hi, I had some problems importing a pcb from Altium (17.0.6.354, not sure but at least 3 years old) .
It's a 15 layer design with ~330 SMD footprints (a 200mm diameter circle).
The converted .kicad_pcb file results in a number of errors, which I corrected in a text editor until Kicad pcbnew (5.1.8) could load the file; for each I copy the relevant lines:
-message: 15 is not a valid layer count in input/source
	...
	(11 In11.Cu signal)
	(12 In12.Cu signal)
	(13 In13.Cu signal)

	(31 B.Cu signal)
	...
	
	Solved by manually adding layer 14 in the blank line above:
	(14 In14.Cu signal)

-message:  Expecting ")" in input/source
	(zone  (layer ) (tstamp 547BA6E6) (hatch edge 0.508)
	Substituted with (layer 14.Cu) 279 times
	
-message:  Expecting ")" in input/source
	(zone (net 414 "SDA_MUX1") (layer F.Cu) (tstamp 547BA6E6) (hatch edge 0.508)
	Substituted similar lines with (14K times): 
	(zone (net 414) (net_name SDA_MUX1) (layer F.Cu) (tstamp 547BA6E6) (hatch edge 0.508)
	
-message:  Expecting ")" in input/source
	(zone (net 941) (net_name VCC uC) (layer B.Cu) (tstamp 547BA6E6) (hatch edge 0.508)
	net_name is split in two words, substituted with single word like VCC_uC (13 times)   

-warning: there is a zone that belongs to a not existing net "VCC_uC" ... and other nets
	Just ignored
	
-In the resulting design, zones have vertices all over the drawing space, up to 1.5m from origin, plus at least a vertex at [0,0]
	I deleted all zones
	
[image]

thesourcerer8, 2021-01-12:
Could you provide the .PcbDoc for analyzing the odd layer number issue? If we count Front and Back layer too, I am counting 16 layers in your design. Due to the missing layer, the layer-references are broken.
I have added some code to remove spaces from net names, I hope this will not cause collissions for someone else.
I am currently working on the zones problem. Could you perhaps try to provide a minimum test-file that shows the errorneous zones with vertex at [0,0] ? The problem only seems to happen with huge files for me.

albmac, 2021-01-13:
Sorry but the .PcbDoc is confidential (and quite big), I cannot share it.
16 layers counting the one I renamed (In14.Cu) but there seems to be nothing attached to it; the comments describe only 15 layers.
Maybe the fastest solution would be an option to ignore fill zones (they are anyways likely to be removed and re-generated automatically in kicad).
I don't have Altium so I unfortunately cannot provide test files.

thesourcerer8, 2021-01-13:
Ok, no problem. I'll let you know when the bugs are fixed.