thesourcerer8/altium2kicad #106: proper test infrastructure

bunker-bunk567, 2021-02-02:
Would it not be possible to use gerber files as reference to verify the correctness of the conversion process? Either for synthetic boards designed to test features comprehensively or for real world boards to test real world sturdiness.

Imagine the persuasive power you could get out of such a proof of correctness. A tool that can be proven to be correct for dozens of third party boards would be a hell of an argument for switching from altium to kicad.

thesourcerer8, 2021-02-02:
I have been using gerber-files for differential testing quite a lot, but there are a number of things that aren't specified in the layout file, that are handled differently by different tools, which results in differences in gerber-files that cannot be avoided. For example the angles of the segments of thermal reliefs for pads. One tool does them 0°, 90°. Another tool does them in 45° angles ...  . Then there are different fonts/kerning/... for the texts... lots of variations that lead to the fact that you cannot guarantee a 100% identical gerber files. So we would still need some mechanism do define which differences are acceptable , and which differences are not acceptable. If anyone could code up a regression testing system for gerber files, which allows to waive such issues, that would be great, yes.

bunker-bunk567, 2021-02-03:
https://imgur.com/a/WTazWcB

first image is black kicad on top of white altium.

second image is the same gerber with the roles reversed.

i think it's obvious there is something useful about gerber diffs.

Look for example at the plenty white dots in the first image. These are sort of deltas that smoothly connect tracks to PTH in altium, whereas kicad just plops the 2 shapes together. The tracks and PTHs have tolerance values already, so you can ignore all those little dots because of properties of the elements they are part of.

bunker-bunk567, 2021-02-03:
also, you can manually OK differences from such a diff, for example the angle problem on thermal reliefs and produce a mask. if you want to keep doing regression tests with gerbers from those layers, you can use the mask to then automatically OK these variations. You can also automatically find differences that have the same shape and compare their coordinates with objects in the layout file and come up with rules and policies.

bunker-bunk567, 2021-02-03:
> For example the angles of the segments of thermal reliefs for pads. One tool does them 0°, 90°. Another tool does them in 45° angles ... . Then there are different fonts/kerning/... for the texts... lots of variations that lead to the fact that you cannot guarantee a 100% identical gerber files.

That's really up to kicad-devel. If we can show them what layout features they need to implement to be 99.9% accurate, i think they will jump at the opportunity.

Minus fonts, but that obviously does not matter.

Minus slight variations due to whatever reason. If a user can see that the maximum width of deviations is 1mil that surely won't matter. At least to the 99% of pedestrians who don't design RF laser scalpels.

bunker-bunk567, 2021-02-04:
@thesourcerer8 any chance to get some help on this?

i need someone with an altium installation to click a few buttons.

(a) produce gerbers for existing altium projects

(b) maybe later place a few chosen sample elements on a board

bunker-bunk567, 2021-02-05:
making some progress:

https://i.imgur.com/RdUtpFI.png