thesourcerer8/altium2kicad #108: Converted Schematic not Showing symbols

BogdanTheGeek, 2021-02-16:
Hi, I have tried to convert a simple project and the pcb looks perfect, but the schematic is missing symbols.
I also have the same issue as mentioned here [https://github.com/thesourcerer8/altium2kicad/issues/101](url).
if you have a hint of what may cause the issues above I can try to make the simplest possible projects that have that issues.
[image]

BogdanTheGeek, 2021-02-16:
I was able to fix this by adding the .lib file to the project symbols libraries and then pressing this and selecting the correct symbol for the few that matched multiple libraries
[image]

thesourcerer8, 2023-06-05:
I just tried the new Altium Importer that has been integrated into Kicad itself (one in PcbNew and on in EESchema, to be found in "File -> Import -> Non-KiCad Files", and the results are much better than altium2kicad, so I would suggest to you use the KiCad integrated importer yourself instead of altium2kicad. If it works for you, please close this issue. If it doesn't work for you, please let me know.