thesourcerer8/altium2kicad #109: Cannot open converted pcb

BogdanTheGeek, 2021-02-16:
I am getting this error:
[image]
here is a snippet of the output file at that section:
[image]

cmb49, 2021-04-16:
I get the same error.

cmb49, 2021-04-16:
Here is what I did:
I created a zip file but with only "one" of the schematic pages, i.e. MCU_144_IOs.SchDoc and the Design.PcbDoc file.  I used the online Altium2kiCad and got the altium2kicad.zip file back. I was able to open the schematic of this one page! (with some errors in some components but most of it is there). I hope this helps. I will do the same for each of the schematic pages.

thesourcerer8, 2023-06-05:
I just tried the new Altium Importer that has been integrated into Kicad itself (one in PcbNew and on in EESchema, to be found in "File -> Import -> Non-KiCad Files", and the results are much better than altium2kicad, so I would suggest to you use the KiCad integrated importer yourself instead of altium2kicad. If it works for you, please close this issue. If it doesn't work for you, please let me know.