thesourcerer8/altium2kicad #111: need a one more bracket

alex-invntr, 2021-03-24:
[image]
[image]
[image]
[image]

alex-invntr, 2021-03-24:
adding a parenthesis removes the error and the file is converted

thesourcerer8, 2021-04-01:
I pushed an addition to prevent too long lines (line #17868 might be too long for KiCad) , please try it whether that solves the problem.

yorickdowne, 2021-04-10:
Seen a very similar issue. This is what it looked like originally:

```
(zone  (layer F.Cu) (tstamp 547BA6E6) (hatch edge 0.508)
    (connect_pads thru_hole_only (clearance 0.09144))
    (fill (mode segment) (arc_segments 32) )
    (polygon
      (pts
(xy 90.82088008 -150.95257846) (xy 91.575570148 -150.893183238) (xy 91.567586416 -150.6899831) (xy 90.3459839 -150.6899831) (xy 90.3459839 -149.2819849) (xy 92.3539809 -149.2819849) (xy 92.3539809 -149.46803228) (xy 93.987860356 -149.46803228) (xy 98.047106876 -145.408828475) (xy 97.83309618 -144.89216132) (xy 99.6882436 -144.12373512) (xy 100.22706142 -145.42455548) (xy 98.979698917 -145.941229259) (xy 97.063295931 -147.85761208) (xy 97.178399349 -148.029877446) (xy 97.30392528 -147.97788286) (xy 97.421207271 -147.962442169) (xy 97.67470432 -147.70894512) (xy 97.82061462 -147.59698192) (xy 97.99053554 -147.52659852) (xy 98.1728796 -147.50259298) (xy 100.154157838 -147.50259298) (xy 100.9510808 -147.17249712) (xy 101.48989862 -148.47331748) (xy 100.230892872 -148.994814043) (xy 100.303738227 -149.184582702) (xy 102.053136 -148.56508768) (xy 103.8284055 -147.82974752) (xy 105.50295666 -147.031028877) (xy 105.420279507 -146.845332677) (xy 104.1617932 -147.36661408) (xy 103.62297538 -146.06579372) (xy 104.30200628 -145.784530286) (xy 104.30200628 -144.608042) (xy 104.34143216 -144.40983056) (xy 104.45371032 -144.24179432) (xy 109.20541151 -139.49009313) (xy 108.86021662 -139.14489824) (xy 110.132642688 -137.872472172) (xy 109.735028072 -137.474851115) (xy 109.531826128 -137.474850292) (xy 109.49238706 -137.51428936) (xy 108.07251944 -136.09442174) (xy 109.35096494 -134.81597624) (xy 110.77083256 -136.23584386) (xy 110.629780308 -136.376896112) (xy 112.89958116 -138.64673566) (xy 113.01154182 -138.7926485) (xy 113.08192268 -138.96256434) (xy 113.10592822 -139.1449084) (xy 113.10592822 -140.55087206) (xy 113.60913778 -141.05408162) (xy 112.902039019 -141.761180381) (xy 112.89957608 -141.76439008) (xy 112.750542092 -141.913424068) (xy 112.888580918 -142.062753634) (xy 113.4590187 -141.57555378) (xy 114.85285386 -140.25285386) (xy 116.17555378 -138.8590187)  (xy 116.667124494 -138.283463297) (xy 116.517794928 -138.145424472) (xy 115.65408162 -139.00913778) (xy 114.65847528 -138.01353144) (xy 116.0783429 -136.59366382) (xy 116.20989966 -136.72522058) (xy 117.37955188 -135.55556836) (xy 117.37955188 -129.85208913) (xy 116.86866136 -129.6456767) (xy 117.6208722 -127.7838948) (xy 117.649217531 -127.795347016) (xy 117.761371382 -127.625900802) (xy 117.487446 -127.35197542) (xy 117.021841798 -127.35197542) (xy 116.80857512 -127.86684866) (xy 116.239601975 -127.631171801) (xy 116.02052742 -127.850258573) (xy 116.02052742 -129.505816367) (xy 116.09253896 -129.5996646) (xy 116.18211714 -129.81592528) (xy 116.2126708 -130.048) (xy 116.18211714 -130.28007472) (xy 116.09253896 -130.4963354) (xy 115.95004242 -130.68204242) (xy 115.7643354 -130.82453896) (xy 115.54807472 -130.91411714) (xy 115.316 -130.9446708) (xy 115.08392528 -130.91411714) (xy 114.8676646 -130.82453896) (xy 114.68195758 -130.68204242) (xy 114.53946104 -130.4963354) (xy 114.44988286 -130.28007472) (xy 114.4193292 -130.048) (xy 114.44988286 -129.81592528) (xy 114.53946104 -129.5996646) (xy 114.61147258 -129.505816367) (xy 114.61147258 -127.5584444) (xy 114.63547812 -127.37610034) (xy 114.674629793 -127.281581051) (xy 114.70585136 -127.2061972) (xy 114.81780948 -127.06028182) (xy 115.076301378 -126.801775507) (xy 115.64531608 -125.42804734) (xy 116.888327583 -125.94292058) (xy 117.77926914 -125.94292058) (xy 117.9616132 -125.96692612) (xy 118.13153412 -126.03730952) (xy 118.27744442 -126.14927272) (xy 121.45541368 -129.32724198) (xy 121.461988513 -129.335810294) (xy 121.98608064 -129.5475565) (xy 121.618822023 -130.456550907) (xy 121.804900183 -130.538427297) (xy 122.42974752 -129.2284055) (xy 123.16508768 -127.453136) (xy 123.80651134 -125.64181374) (xy 123.870429608 -125.426029457) (xy 123.678369418 -125.358958907) (xy 123.0694262 -126.866142) (xy 121.76395256 -126.3386983) (xy 122.041221443 -125.652435672) (xy 120.919775973 -124.530952394) (xy 0 0) (xy 0 0) # Verts: 51 
```

There are three closing brackets missing. This should end in
```
(xy 0 0) )
 )
)
# Verts: 51
```

I saw a good chunk of "Use of uninitialized value in division" and "Use of uninitialized value in unpack" during conversion. The missing brackets might just be a symptom, and the root cause is something entirely different.

Let me know what I can do to help hunt down root  cause.

thesourcerer8, 2023-06-05:
I just tried the new Altium Importer that has been integrated into Kicad itself (one in PcbNew and on in EESchema, to be found in "File -> Import -> Non-KiCad Files", and the results are much better than altium2kicad, so I would suggest to you use the KiCad integrated importer yourself instead of altium2kicad. If it works for you, please close this issue. If it doesn't work for you, please let me know.