thesourcerer8/altium2kicad #114: `"` symbol in input file needs to be escaped in output

yorickdowne, 2021-04-10:
This KiCAD output because of a `.2" SCREW TERMINAL` reference:

```
(module ".2" SCREW TERMINAL" (layer F.Cu) (tedit 4289BEAB) (tstamp 539EEDBF)
```

Won't load because the `.2"` closes the opening `"`

Escaping the character works:

```
(module ".2\" SCREW TERMINAL" (layer F.Cu) (tedit 4289BEAB) (tstamp 539EEDBF)
```

thesourcerer8, 2021-04-12:
Problem has been fixed, please update and try again.

yorickdowne, 2021-04-12:
That fixed it, thank you!

I piled unto #111 with an issue where two `pts` are `0, 0`, and shouldn't be. Closing brackets are missing, and conversion shows errors at that point. I'll give you a new issue for it rather than clutter 111 further.

thesourcerer8, 2021-04-12:
Thanks for the feedback!

closed 2021-04-12 by thesourcerer8