thesourcerer8/altium2kicad #115: Issue converting EasyEDA's files using  "export as Altium Designer"

morakniv, 2021-04-11:
I get error code: 
"File format not understood. Please upload ZIP files which contain .PcbDoc or .SchDoc files. " 

**EasyEDA ->  Export as Altium -> .ZIP  -> A2K** 

This is the file extension(s) I get from EasyEDA: 
Arduino VU meter skit_2021-04-10.pcbdoc - 1254 kB
Arduino VU meter skit_2021-04-10.schdoc  - 310 kB

I tried converting both PCB/SCH files together and one at a time, always inside a ZIP folder.  
Does EasyEDA differ from files generated by Altium Designer? Or have I missed something in the readme?

arch3angel, 2022-05-11:
I have this same issue.  Did you ever figure out the issue

bestel74, 2022-06-14:
Same here, maybe someone has a solution?

thesourcerer8, 2022-06-14:
Hmm, you could try to rename the files to .PcbDoc if you are running on a case-sensitive filesystem. If that doesn't help, can you provide me some test-files please?

bestel74, 2022-06-14:
I used the online tool (and tried to rename) but same issue.
I can't share my files, but here is the log output if this can help:

> Archive:  db/7961.zip
  inflating: db/7961/Acquisition_v1.PcbDoc  
  inflating: db/7961/Acquisition_v1.SchDoc  
Loading Acquisition_v1.SchDoc
Loading Acquisition_v1.PcbDoc
Skipping ASCII .PcbDoc
Done.
Handling Acquisition_v1.SchDoc
No unpacked .PcbDoc documents found. Trying to unpack it:
Loading Acquisition_v1.SchDoc
Loading Acquisition_v1.PcbDoc
Skipping ASCII .PcbDoc
Done.
There were no unpacked PcbDoc files found.

thesourcerer8, 2022-06-17:
Ahh, those files are in the Protel ASCII format, not in the Altium Binary format. You could convert them to the binary format if you had an Altium license, and then use my converter. I had been working on improving the converter to be able to handle the ASCII format in the past, but it does not seem to work at the moment, I will look into that again.

bestel74, 2022-06-17:
Ok, thanks for the reply :)

arch3angel, 2022-08-30:
> Ahh, those files are in the Protel ASCII format, not in the Altium Binary format. You could convert them to the binary format if you had an Altium license, and then use my converter. I had been [...]

Thank you for looking at a possible solution!  This is very frustrating, I like the ease of using EasyEDA with the parts and JLCPCB integration, but I don't want my work locked to a software platform.  I would rather store them as a back up in a general format so anyone can use them.  Anything you can do would be a huge help!