thesourcerer8/altium2kicad #116: Cannot open converted schematic from Altium 15

PigHillJimster, 2021-04-12:
Cannot open converted schematic from Altium 15 saved in Altium ascii format. Error from Kicad attached. Conversion log attached.

Is there a private area to submit the Altium file, please?

[altium2kicad-conversionlog.txt](https://github.com/thesourcerer8/altium2kicad/files/6295007/altium2kicad-conversionlog.txt)

[altium2kicad_errors_opening_converted_design_log.txt](https://github.com/thesourcerer8/altium2kicad/files/6294992/altium2kicad_errors_opening_converted_design_log.txt)

thesourcerer8, 2021-04-12:
Please see my message on LinkedIn

thesourcerer8, 2021-04-12:
Ah, I found the problem. You tried to open the file directly with kicad, and kicad tries to interpret it as a project file. You should open it with eeschema, that should work.

PigHillJimster, 2021-04-14:
This issue is now closed. Thank you.

closed 2021-04-14 by PigHillJimster