thesourcerer8/altium2kicad #122: No .pro file generated

eagarciaj, 2021-11-28:
I open FetchUpSTel-PcbDoc.kicad_pcb and generate the pro file but a message error openning the sch
"Cannot open library PoeSTel-SchDoc-cache.lib" print screen attached,
[image]
[altium2kicad.zip](https://github.com/thesourcerer8/altium2kicad/files/7612831/altium2kicad.zip)

Altium files attache.

Please advice
Best Regards,

thesourcerer8, 2022-11-24:
Which KiCad version do you use?

eagarciaj, 2022-11-24:
Kicad v6.0.9

thesourcerer8, 2023-06-05:
I just tried the new Altium Importer that has been integrated into Kicad itself (one in PcbNew and on in EESchema, to be found in "File -> Import -> Non-KiCad Files", and the results are much better than altium2kicad, so I would suggest to you use the KiCad integrated importer yourself instead of altium2kicad. If it works for you, please close this issue. If it doesn't work for you, please let me know.