thesourcerer8/altium2kicad #123: No .pro file generated

tginin, 2022-01-06:
Hello team,
After the uploading of Altium zip archive and the downloading altium2kicad.zip, I don't find *.pro file: 

Archive:  db/5688.zip
  inflating: db/5688/AW9253_TT.png   
  inflating: db/5688/InoxSysLogo.png  
  inflating: db/5688/Piezo_ADC.SchDoc  
  inflating: db/5688/Piezo_LED.SchDoc  
  inflating: db/5688/Piezo_PWR.SchDoc  
  inflating: db/5688/Piezo_SEN.PcbDoc  
  inflating: db/5688/PT100-4.png     
  inflating: db/5688/PT100.png       
  inflating: db/5688/tPLH.png        
Loading Piezo_ADC.SchDoc

Loading Piezo_LED.SchDoc

Loading Piezo_PWR.SchDoc

Loading Piezo_SEN.PcbDoc

Done.
Handling Piezo_ADC-SchDoc/Root Entry/FileHeader.dat
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
This circle/arc is not part of a component, but KiCad does not support that. As a workaround we are creating a dummy component.
Unhandled Record type within: 34
Unhandled Record type within: 34
This could be oval or rounded rectangle
This could be oval or rounded rectangle
This could be oval or rounded rectangle
This could be oval or rounded rectangle
This could be oval or rounded rectangle
Handling Piezo_LED-SchDoc/Root Entry/FileHeader.dat
WARNING: Pin symbol type 11 not understood - IGNORING!
WARNING: Pin symbol type 34 not understood - IGNORING!
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
Unhandled Record type within: 34
Unhandled Record type within: 34
This circle/arc is not part of a component, but KiCad does not support that. As a workaround we are creating a dummy component.
This circle/arc is not part of a component, but KiCad does not support that. As a workaround we are creating a dummy component.
This circle/arc is not part of a component, but KiCad does not support that. As a workaround we are creating a dummy component.
This circle/arc is not part of a component, but KiCad does not support that. As a workaround we are creating a dummy component.
Unhandled Record type within: 34
This could be oval or rounded rectangle
This could be oval or rounded rectangle
This could be oval or rounded rectangle
This could be oval or rounded rectangle
This could be oval or rounded rectangle
This could be oval or rounded rectangle
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
This could be oval or rounded rectangle
This could be oval or rounded rectangle
Handling Piezo_PWR-SchDoc/Root Entry/FileHeader.dat
Unhandled Record type without: 7  (#2)
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
Unhandled Record type within: 34
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
Handling Piezo_SEN-PcbDoc/Root Entry/FileHeader.dat
Handling Piezo_ADC.SchDoc
Handling Piezo_LED.SchDoc
Handling Piezo_PWR.SchDoc
Handling Piezo_SEN-PcbDoc/Root Entry/Board6/Data.dat
ASCII-Piezo_SEN-PcbDoc.PcbDoc exists?
Writing to Piezo_SEN-PcbDoc/Root Entry/Board6/Data.dat.txt
Writing to Piezo_SEN-PcbDoc/Root Entry/Rules6/Data.dat.txt
Writing to Piezo_SEN-PcbDoc/Root Entry/Classes6/Data.dat.txt
Writing PCB to Piezo_SEN-PcbDoc.kicad_pcb
Pads6...
Writing to Piezo_SEN-PcbDoc/Root Entry/FileVersionInfo/Data.dat.txt
Texts6...

mhanuel26, 2022-06-03:
Hi, 

Same here,

```
Archive:  db/7501.zip
  inflating: db/7501/digital_inputs.SchDoc  
  inflating: db/7501/digital_outputs.SchDoc  
  inflating: db/7501/PCB1.PcbDoc     
  inflating: db/7501/plc_base.SchDoc  
Loading digital_inputs.SchDoc

Loading digital_outputs.SchDoc

Loading plc_base.SchDoc

Loading PCB1.PcbDoc

Done.
Handling digital_inputs-SchDoc/Root Entry/FileHeader.dat
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
Handling digital_outputs-SchDoc/Root Entry/FileHeader.dat
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
Handling PCB1-PcbDoc/Root Entry/FileHeader.dat
Handling plc_base-SchDoc/Root Entry/FileHeader.dat
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
WARNING: Port orientation may be incorrect and thus unconnected - ports on the 'left' of wires may need moving and orientation flipping
Handling digital_inputs.SchDoc
Handling digital_outputs.SchDoc
Handling plc_base.SchDoc
Handling PCB1-PcbDoc/Root Entry/Board6/Data.dat
ASCII-PCB1-PcbDoc.PcbDoc exists?
Writing to PCB1-PcbDoc/Root Entry/Board6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Models/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/UniqueIDPrimitiveInformation/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Rules6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Classes6/Data.dat.txt
Writing PCB to PCB1-PcbDoc.kicad_pcb
Writing to PCB1-PcbDoc/Root Entry/Nets6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Components6/Data.dat.txt
Pads6...
Seems we should skip 0x1e0 bytes
Seems we should skip 0x1e0 bytes
[... 42 lines cut ...]
NOT FOUND: HSMY-C197.wrl
[... cut ...]
Writing to PCB1-PcbDoc/Root Entry/ShapeBasedComponentBodies6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Components6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Arcs6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Vias6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Polygons6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Tracks6/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/FileVersionInfo/Data.dat.txt
Writing to PCB1-PcbDoc/Root Entry/Regions6/Data.dat.txt
Overflow!
Overflow!
Overflow!
Texts6...
```

mistyron, 2022-09-03:
Yes, same here but I can still investigate pcb and schematic files

thesourcerer8, 2022-11-24:
I have added convertall.pl now, which converts all and creates a .kicad_pro file now.