thesourcerer8/altium2kicad #126: Convert 'gr_arc' to 'segment' with Python

apeng2012, 2022-05-07:
After converting with altium2kicad there may be 'gr_arc' in the Cu layer. 'gr_arc' has no net attributes

Can it be added to the project to make the 'segment' net complete

```python
import re
import numpy as np

def arc2segment(center, point, angle):
    ret = []
    vector = np.array([point[0] - center[0], point[1] - center[1]])
    alpha = np.deg2rad(angle)
    step = np.deg2rad(10) * np.sign(alpha) * -1
    angles = np.arange(alpha, 0, step)
    for i in angles[::-1]:
        x = vector[1]*np.sin(i*-1) + vector[0]*np.cos(i) + center[0]
        y = vector[1]*np.cos(i) + vector[0]*np.sin(i) + center[1]
        ret.append([x, y])

    return ret

fw = open('output.kicad_pcb', 'w', encoding='utf8')

p = re.compile(r' *[(]gr_arc [(]start ([+-]?[\d\.]*) ([+-]?[\d\.]*)[)] [(]end ([+-]?[\d\.]*) ([+-]?[\d\.]*)[)] [(]angle ([+-]?[\d\.]*)[)] [(]layer ([a-zA-Z\.]*)[)] [(]width ([\d\.]*)[)]')

for line in open('input.kicad_pcb', 'r', encoding='utf8'):
    result = p.match(line)

    if result is not None and ('F.Cu' in result.group(6) or 'B.Cu' in result.group(6)):

        center = [float(result.group(1)), float(result.group(2))]
        point = [float(result.group(3)), float(result.group(4))]
        angle = float(result.group(5))
        width = float(result.group(7))
        plist = arc2segment(center, point, angle)
        last = point
        for pt in plist:
            fw.write('\r\n  (segment (start {:.4f} {:.4f}) (end {:.4f} {:.4f}) (width {}) (layer {}) (net 0))'.format(last[0], last[1], pt[0], pt[1], width, result.group(6)))
            last = pt
        continue
    fw.write(line)

fw.close()

```

thesourcerer8, 2022-05-16:
Could you provide annotated screenshots to explain what you mean?

thesourcerer8, 2022-05-16:
Is the problem that the net isn't assigned the correct net? Or is a closed polygon missing a segment?

apeng2012, 2022-05-18:
There is an error in the above program
'(net 0)' changed to '(net 1)'
```python
fw.write('\r\n  (segment (start {:.4f} {:.4f}) (end {:.4f} {:.4f}) (width {}) (layer {}) (net 1))'.format(last[0], last[1], pt[0], pt[1], width, result.group(6)))
```

apeng2012, 2022-05-18:
[https://github.com/SphinxEVK/RT1052V2/blob/master/RT1052EVAbkup.PcbDoc](https://github.com/SphinxEVK/RT1052V2/blob/master/RT1052EVAbkup.PcbDoc)

[image]

···
(gr_arc (start 52.41544 -56.07304) (end 52.57800 -56.07304) (angle -90) (layer F.Cu) (width 0.1016))
···

Convert 'gr_arc' to many more small 'segment'

```
  (segment (start 52.5682 -56.1286) (end 52.5562 -56.1543) (width 0.1016) (layer F.Cu) (net 1))

  (segment (start 52.5562 -56.1543) (end 52.5400 -56.1775) (width 0.1016) (layer F.Cu) (net 1))

  (segment (start 52.5400 -56.1775) (end 52.5199 -56.1976) (width 0.1016) (layer F.Cu) (net 1))

  (segment (start 52.5199 -56.1976) (end 52.4967 -56.2138) (width 0.1016) (layer F.Cu) (net 1))

  (segment (start 52.4967 -56.2138) (end 52.4710 -56.2258) (width 0.1016) (layer F.Cu) (net 1))

  (segment (start 52.4710 -56.2258) (end 52.4437 -56.2331) (width 0.1016) (layer F.Cu) (net 1))

  (segment (start 52.4437 -56.2331) (end 52.4154 -56.2356) (width 0.1016) (layer F.Cu) (net 1))
```
[image]

thesourcerer8, 2022-05-20:
I understand the problem now, KiCad was not able to have arcs with nets, so I used graphical arcs instead. I will change the behaviour of the converter to use several segments instead. In parallel I am in contact with the KiCad developers for a better method.

thesourcerer8, 2022-05-23:
KiCad6 provides an arc track, do you want a solution for KiCad 4-5 or KiCad6 ?

thesourcerer8, 2022-06-01:
I have switched the code to target KiCad6 now and implemented the arc's properly now. Please try again now.

closed 2022-06-07 by thesourcerer8