thesourcerer8/altium2kicad #127: altium2kicad creates invalid .kicad_pcb file

BreakingSiam, 2022-07-01:
After running altium2kicad on a large Altium project, it creates a .kicad_pcb file which Kicad version "6.0.4-6f826c9f35~116~ubuntu18.04.1, release build" reports "does not appear to be a valid KiCad project file". 

The generated .kicad_pcb file is 13891785 bytes long, and on visual inspection of the beginning of the file appears to be a valid KiCad project file.

During conversion of the altium project, after displaying writing to "Regions6/Data.dat.txt", altium2kicad reported:

Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Error: substring out of range: length:509 pos:18799 wantedsize:2 field:len[2] (main,../convertpcb.pl:2901 - main::msubstr)
Use of uninitialized value in unpack at ../convertpcb.pl line 2901.
...

The complete conversion output is attached.

[oo.txt](https://github.com/thesourcerer8/altium2kicad/files/9029142/oo.txt)

thesourcerer8, 2022-08-17:
Please provide me with your altium file, so that I can analyze the problem. I couldn't reach you through email.

bjornfor, 2023-06-05:
Has the integrated imprter in KiCAD superseded this project?

thesourcerer8, 2023-06-13:
Yes, it was developed based on this converter, and it seems to have surpassed the quality of this converter now. So unless anyone still needs this converter, I would declare it superseded now.

dzid26, 2023-08-20:
> Yes, it was developed based on this converter, and it seems to have surpassed the quality of this converter now. So unless anyone still needs this converter, I would declare it superseded now.

It works better indeed. Although, for reference, it is in nightly builds. 
And for me, it only worked after first exporting to Circuit Maker / Studio. 
https://forum.kicad.info/t/trouble-importing-pcb-files-from-solidworks-pcb-altium/41689/10?u=dzid26