thesourcerer8/altium2kicad #130: Overflow and No .pro fle

BewilderedAgain, 2022-11-20:
Unable to process this project https://github.com/ketszim97/NearField_PCB_Probes

Archive:  db/14519.zip
25c0d6bc842eb8e2eddc6511ba93796cdc272b2a
  inflating: db/14519/.gitignore     
  inflating: db/14519/NearField_E_Probe.SchDoc  
  inflating: db/14519/NearField_H_Probe.SchDoc  
  inflating: db/14519/NearField_PCB_Probes.PcbDoc  
  inflating: db/14519/NearField_PCB_Probes.PrjPcb  
 extracting: db/14519/NearField_PCB_Probes.PrjPcbStructure  
  inflating: db/14519/NearField_PCB_Probes_BOM.xlsx  
  inflating: db/14519/NearField_PCB_Probes-macro.APR_LIB  
  inflating: db/14519/NearField_PCB_Probes.DRR  
  inflating: db/14519/NearField_PCB_Probes.EXTREP  
  inflating: db/14519/NearField_PCB_Probes.G1  
  inflating: db/14519/NearField_PCB_Probes.G2  
  inflating: db/14519/NearField_PCB_Probes.GBL  
  inflating: db/14519/NearField_PCB_Probes.GBO  
  inflating: db/14519/NearField_PCB_Probes.GBP  
  inflating: db/14519/NearField_PCB_Probes.GBS  
  inflating: db/14519/NearField_PCB_Probes.GKO  
  inflating: db/14519/NearField_PCB_Probes.GM1  
  inflating: db/14519/NearField_PCB_Probes.GM13  
  inflating: db/14519/NearField_PCB_Probes.GM15  
  inflating: db/14519/NearField_PCB_Probes.GM16  
  inflating: db/14519/NearField_PCB_Probes.GM3  
  inflating: db/14519/NearField_PCB_Probes.GM5  
  inflating: db/14519/NearField_PCB_Probes.GPB  
  inflating: db/14519/NearField_PCB_Probes.GPT  
  inflating: db/14519/NearField_PCB_Probes.GTL  
  inflating: db/14519/NearField_PCB_Probes.GTO  
  inflating: db/14519/NearField_PCB_Probes.GTP  
  inflating: db/14519/NearField_PCB_Probes.GTS  
  inflating: db/14519/NearField_PCB_Probes.LDP  
  inflating: db/14519/NearField_PCB_Probes.REP  
  inflating: db/14519/NearField_PCB_Probes.RUL  
  inflating: db/14519/NearField_PCB_Probes.TXT  
  inflating: db/14519/NearField_PCB_Probes.apr  
  inflating: db/14519/NearField_PCB_Probes.zip  
  inflating: db/14519/Status Report.Txt  
  inflating: db/14519/E-FIELD.zip    
  inflating: db/14519/H-FIELD.zip    
  inflating: db/14519/README.md      
  inflating: db/14519/Fusion_Render_1.png  
 extracting: db/14519/Fusion_Render_2.png  
 extracting: db/14519/Fusion_Render_3.png  
  inflating: db/14519/Fusion_Render_4.png  
 extracting: db/14519/Fusion_Render_5.png  
 extracting: db/14519/Fusion_Render_6.png  
Loading NearField_E_Probe.SchDoc

Loading NearField_H_Probe.SchDoc

Loading NearField_PCB_Probes.PcbDoc

Done.
Handling NearField_E_Probe-SchDoc/Root Entry/FileHeader.dat
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
Handling NearField_H_Probe-SchDoc/Root Entry/FileHeader.dat
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
WARNING: Elliptical arcs are not supported in KiCad - creating circular arc using average radius instead
This circle/arc is not part of a component, but KiCad does not support that. As a workaround we are creating a dummy component.
Handling NearField_PCB_Probes-PcbDoc/Root Entry/FileHeader.dat
Handling NearField_E_Probe.SchDoc
Handling NearField_H_Probe.SchDoc
Handling NearField_PCB_Probes-PcbDoc/Root Entry/Board6/Data.dat
ASCII-NearField_PCB_Probes-PcbDoc.PcbDoc exists?
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Board6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Models/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/UniqueIDPrimitiveInformation/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Rules6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Classes6/Data.dat.txt
Writing PCB to NearField_PCB_Probes-PcbDoc.kicad_pcb
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Nets6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Components6/Data.dat.txt
Pads6...
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/ComponentBodies6/Data.dat.txt
NOT FOUND: Conn_Taoglas_Limited_EMPCB_SMAFSTJ_B_HT_eec.wrl
NOT FOUND: CAP0402.wrl
NOT FOUND: Pin Header 1x2 TH Pitch 2.54mm.wrl
NOT FOUND: SOT-89.wrl
NOT FOUND: LQW15AN_W05.wrl
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/ShapeBasedComponentBodies6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Components6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Arcs6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Vias6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Polygons6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Tracks6/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/FileVersionInfo/Data.dat.txt
Writing to NearField_PCB_Probes-PcbDoc/Root Entry/Regions6/Data.dat.txt
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Overflow!
Texts6...

thesourcerer8, 2022-11-24:
I have fixed one problem with the boolean conversion now, and made the Overflows more verbose, but I couldn't fix the Overflows. And I have added a convertall.pl which also generates the .kicad_pro file now, so it is easy to convert and open a whole project now.

thesourcerer8, 2023-06-05:
I just tried the new Altium Importer that has been integrated into Kicad itself (one in PcbNew and on in EESchema, to be found in "File -> Import -> Non-KiCad Files", and the results are much better than altium2kicad, so I would suggest to you use the KiCad integrated importer yourself instead of altium2kicad. If it works for you, please close this issue. If it doesn't work for you, please let me know.