thesourcerer8/altium2kicad #133: "There were no unpacked PcbDoc files found" when unpacking CMPcbDoc file

sjnaylor59, 2023-05-05:
I'm unsure if this is due to a CM update, but I'm unable to convert my layout to kicad. The schematic converts successfully, but the converter can't find the PCB file

[image]
[image]

thesourcerer8, 2023-05-06:
I haven't seen .CMPcbDoc files before, how did you generate that? Try renaming it to .PcbDoc and run the converter and please let me know what happened, and if possible please provide me with the .CMPcbDoc file for analysis.

sjnaylor59, 2023-05-06:
CMPcbDocs are the new way that Circuit Maker strong arms you into sticking with them and moving to Altium quicker. They're feature limited and stops you from bringing Altium PCBs into Circuit Maker.

The same thing happened when I renamed the file to .PcbDoc, it unpacks it (looks a little messy) and then can't find anything to convert using the converter. The files are attatched.

thesourcerer8, 2023-05-06:
I don't see the files attached, perhaps you have to upload them on the Github webinterface?

sjnaylor59, 2023-05-06:
Apologies, take two!
[capacitance doubler circuit - export version.zip](https://github.com/thesourcerer8/altium2kicad/files/11413454/capacitance.doubler.circuit.-.export.version.zip)