thesourcerer8/altium2kicad #134: Cannot convert Scheme and Converting PCB fails

highvoltageenjoyer, 2023-06-01:
So I have a project in our lab which is completely made withing Altium and want to convert it into KiCAD.

With the conversion of the Schematic I get this error code:

Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 867.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 867.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 868.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 869.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 867.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 869.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 869.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 870.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 867.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 867.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 868.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 868.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 868.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 869.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 869.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 869.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 870.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 870.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 870.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 870.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 867.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 867.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 867.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 868.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 868.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 868.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 868.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 869.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 869.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 870.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 870.
Use of uninitialized value in multiplication (*) at /home/oliver/Programs/altium2kicad/convertschema.pl line 870.

Where at this point the conversion fails completely and I do not get a KiCAD schematic file.

With the PCB conversion the error starts when I try to open KiCAD with a new Project and then append the converted pcb. It gives me an overflow error in such that I can't open the pcb file in any way.

It would be great if you could help us in our lab with this. Since we only have one computer with an Altium license I want to get as much independent as possible through KiCAD.

In the attachment you will find the Project files.
[svd.zip](https://github.com/thesourcerer8/altium2kicad/files/11626332/svd.zip)

highvoltageenjoyer, 2023-06-02:
Thank you for the response! 
Yea sure! You'll find it in the Attachments.
I kind of have found out at when it stopped working, it seems like there was a new KiCAD update recently. This maybe caused things to not work anymore.
[svd_single_channel_revB.pdf](https://github.com/thesourcerer8/altium2kicad/files/11635387/svd_single_channel_revB.pdf)

highvoltageenjoyer, 2023-06-12:
So the actual version 7.0 of KiCAD does not feature a conversion vom altium to KiCAD, the same with the nightly version which I had to compile myself. This accounts for Linux. It seems like, or as far as I have seen it, this kind of conversion only exists on KiCAD versions for Windows. 

So sadly this problem still exists :(

thesourcerer8, 2023-06-13:
It also works the same way in 7.0.5. 
For PCB Layout design, you can use the kicad tool, create a new project, go the the layout editor, click on File -> Import -> Non-KiCad-board file.
For Schematic design, I suggest to start eeschema directly (not to go through the kicad project management tool), -> File -> Import -> Non-KiCad-plan.
[image]

highvoltageenjoyer, 2023-06-13:
Omg it finally worked!
So the working process is like this:

Do not open a Non-KiCAD Projekt but JUST the PCB and Schematic. That was the fault I did all the time.
So you directly go to the PCB and Schematics Editor and there you import the Non-KiCAD File.

Thank you very much!

highvoltageenjoyer, 2023-06-14:
Yes we can :)

closed 2023-06-14 by highvoltageenjoyer