thesourcerer8/altium2kicad #135: Unhandled exception class: class std::bad_function_call

mizrex, 2023-06-01:
Thank you for the wonderful tool. I appreciate any help with this.
Unfortunately, I'm running into this issue, where the file generated is throwing this error: 
I have tried zipping the files, removing some sheets. But it gives the same error every time.
[image]

Below is the screenshot of the Altium files and their respective extensions.

[image]

thesourcerer8, 2023-06-01:
Could you please provide all the files? I haven't seen that error before.
Have you run the convertall.pl ?

thesourcerer8, 2023-06-04:
I slightly improved the conversion, but I see that there are far more improvements necessary. Could you please provide PDFs of the Schematics from Altium?

thesourcerer8, 2023-06-05:
I tried again now, and couldn't reproduce the error message you have written initially, I guess it might be a bug that will get fixed in the next KiCad version, or you might want to report it to KiCad yourself. I just tried the new Altium Importer that has been integrated into Kicad itself (one in PcbNew and on in EESchema, to be found in "File -> Import -> Non-KiCad Files", and the results are much better than altium2kicad, so I would suggest to you use the KiCad integrated importer yourself instead of altium2kicad. If it works for you, please close this issue. If it doesn't work for you, please let me know.

mizrex, 2023-06-05:
Thank you for getting back and trying. Yes, the KiCad importer worked great. 
Appreciate your help with this.